OpenHantek/OpenHantek6022 #268: Hantek IDSO1070

polluks, 2022-02-06:
How about to support this model?
https://github.com/hhornbacher/idso1070-protocol

Ho-Ro, 2022-02-17:
I can only develop and test with available HW. Even these scopes look the same as 6022 they are totally different from internal structure: 6022 samples directly to USB without internal buffer while they buffer internally and transmit later at lower rate.
I stripped down "OpenHantek6022" from "openhantek" to support only the 6022 models due to the lack of available HW. "openhantek" supported also these kind of buffered scopes, but unfortunately it is unmaintained since 2019.

Ho-Ro, 2022-03-14:
I will close for now, we can reopen if someone is willing to help.

closed 2022-03-14 by Ho-Ro