OpenHantek/OpenHantek6022 #279: Quick right-click measurement with on-screen display in demo mode does not work correctly.

spam-receiver, 2022-03-25:
It is cleared at every refresh of graph.

[image]

ericfont, 2022-03-25:
It works for me in demo mode.  I don't think your issue is specific to *demo mode*. I think your issue would happen in regular mode too. Your issue is probably unrelated to demo mode, but probably more likely something with your system setup. We could provide better assistance if you provide more information, such as your Qt version, mesa version, OS, and the other bits of information listed in the bug report:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
**No support for non-Linux related issues unless a volunteer steps in!**

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device [e.g. *Hantek 6022BE, ...*]
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Ho-Ro, 2022-03-25:
Cannot reproduce - as eric wrote, please provide more info, e.g. the start message when called from cmd.exe or the output of `Help/About`, especially this line
`Graphic: 2.1 Mesa 20.3.5 - GLSL version 1.20`

> Windows 10 Pro x64 19044.1586
Nvidia GTX 1050

I always mistrust Nvidia (as well as Windows) ;)

Please try the command line options 
```
  -e, --useGLES                Use OpenGL ES instead of OpenGL
  --useGLSL120                 Force OpenGL SL version 1.20
  --useGLSL150                 Force OpenGL SL version 1.50

```

spam-receiver, 2022-03-25:
"--useGLSL120" or " --useGLSL150" - all the same.

"-e" or "--useGLES"  - blank screen instead of graph screen.

Command line without parameters:

```
OpenHantek6022 (20220325 - commit 976)
"4.6.0 NVIDIA 512.15"
Graphic: 4.6.0 NVIDIA 512.15 - GLSL version 1.20
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
```

with "-e" or "--useGLES":

```
OpenHantek6022 (20220325 - commit 976)
"4.6.0 NVIDIA 512.15"
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
QOpenGLWidget: Failed to create context
QPainter::begin: Paint device returned engine == 0, type: 1
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::font: Painter not active
QPainter::setFont: Painter not active
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
-//-
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
```

with "--useGLSL120":

```
OpenHantek6022 (20220325 - commit 976)
"4.6.0 NVIDIA 512.15"
Graphic: 4.6.0 NVIDIA 512.15 - GLSL version 1.20
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
```

with "--useGLSL150":

```
OpenHantek6022 (20220325 - commit 976)
"4.6.0 NVIDIA 512.15"
Graphic: 4.6.0 NVIDIA 512.15 - GLSL version 1.50
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
```

spam-receiver, 2022-03-26:
No problems under KDE Plasma in Fedora 35 Linux with the same hardware.

Ho-Ro, 2022-04-11:
#289 mentions also some issues with NVidia OpenGL 4.6.0:
> Graphic: 4.6.0 NVIDIA 510.54 - GLSL version 1.20

Ho-Ro, 2022-05-31:
As nobody found a solution I will close for now, please reopen if necessary.

closed 2022-05-31 by Ho-Ro