OpenHantek/OpenHantek6022 #281: tooltips not showing in menu

ericfont, 2022-03-29:
**Describe the bug**
If I hover mouse over a top menu or over submenu, I don't see any of the tooltips, even though I know in the code that there are tooltips.

**Computer environment (please complete the following information):**
 - OpenHantek version: commit 977
 - OS: arch linux rolling
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Device: DSO-6022BL (0664E42F01E8), FW0210
 - Graphic: 4.6 (Compatibility Profile) Mesa 21.3.7 - GLSL version 1.20
 - Qt version: 5.15.3

ericfont, 2022-03-29:
I can see tooltips on the left-column of buttons.  But I don't see tooltips on the top-menu, nor do I see tooltips on the right-panel.

Ho-Ro, 2022-03-30:
Thx for reporting. Menu tooltips are not visible unless enabled with [`setToolTipsVisible()`](https://doc.qt.io/qt-5/qmenu.html#toolTipsVisible-prop). I will update accordingly soon.

Ho-Ro, 2022-04-04:
Fixed during my big refactoring [2fb76ba](https://github.com/OpenHantek/OpenHantek6022/commit/2fb76bae4e133349d683c99fcedda250484eafc8).

https://github.com/OpenHantek/OpenHantek6022/blob/5973a32a0201af8caf9262f3f26c77a6d73b0d1c/openhantek/src/mainwindow.cpp#L168
```
    ui->menuFile->setToolTipsVisible( true );
    ui->menuExport->setToolTipsVisible( true );
    ui->menuView->setToolTipsVisible( true );
    ui->menuOscilloscope->setToolTipsVisible( true );
    ui->menuHelp->setToolTipsVisible( true );
```

closed 2022-04-04 by Ho-Ro