OpenHantek/OpenHantek6022 #284: Tooltips

spam-receiver, 2022-04-05:
Configuration option to enable/disable tooltips.

Ho-Ro, 2022-04-05:
Sure, this is already quite high on my feature list, the tootip thing was just a proof of concept. I also like it more when an UI is clean and silent, but I learned from a lot of beginner questions that people do not read the manual and are sometimes clueless in front of these complex machines.

spam-receiver, 2022-04-05:
I sincerely thank you for all your creative efforts.

P.S.: I'll try to change the source of the calibration signal from PC2 to PE2 on my DSO-6022BL and let you know about the results.

Ho-Ro, 2022-04-05:
The signal at PE2 is a small pulse when the timer T2 overflows (PC2 is toggled by the interrupt service routine) - you must put a toggle flip-flop (e.g. 74HC74, connect PE2 -> Clk and /Q -> D) behind to get the right frequency and 50% duty cycle.
[image]

commit 19a93a99 (2022-04-05): Close #284 - add config dialog: show/hide tooltips for user interface