OpenHantek/OpenHantek6022 #285: Save Settings for x-position of trigger not restored when loading the settings file.

ericfont, 2022-04-06:
**Describe the bug** / **To Reproduce**
* Change the x position of a channel's trigger.
* Goto "File"->"Save Settings as..." and save the settings file.
* Change the x position of that channel's trigger.
* Goto "File"->"Open Settings" and open that saved settings file.
* Observe that the x position of the trigger reverts back to the left edge.

**Expected behavior**
Opening that saved settings file should have restored the previously-saved x position of the channel's trigger.

**Computer environment (please complete the following information):**
 - OpenHantek version: 20220404 - commit 981
 - OS: linux 5.15.32-1-lts
 - Distribution, version: Arch linux x86-64 rolling
 - Video hardware: Radeon
 - OpenGL version: Mesa 22.0.0 GLSL version 1.20
 - Qt version: *Qt 5.15.3*
 - C++ compiler version: *gcc 11.2.0

ericfont, 2022-04-06:
Funny thing is this failure to save & restore seems to mainly only happen if the only setting I change before saving is the x-position of the trigger.  But if I change other settings, like the y-position of the trigger and the y-offset of the channel as well as the x-position of the trigger, then upon saving those changes and making other changes and reloading, then those earlier changes are correct restored.  And if I start a fresh session, I can restore that settings file that had multiple things changed fine.  But if the only thing changed for the settings file was the x-position of the trigger, then that x-position of the trigger is not restored.

Ho-Ro, 2022-04-06:
The reason for this illogical behaviour lies in the handling of the persistent storage - a heritage of the predecessor project *openhantek*. In contrast to the statement in the menu text `Save as ..`, it is interpreted internally as *"Change the persistent storage location for the current session from default to .."* and then saves the current status. So if you change something, save it *"as .."* and undo the change, the previously saved file is overwritten. The same applies if you select `Open settings ..`, it also does *"Change permanent location from default to .."* and then loads the settings from this file, any further changes will also change this file again. 
I have rearranged the saving of the settings to solve the problem, `Save as ..` now saves the current state to the specified file but keeps the default location; likewise `Open settings ..`. In addition, a config file can now be specified as a command line parameter `OpenHantek -c <CONFIGFILE>`, this also keeps the default persistent location unchanged.

closed 2022-04-06 by Ho-Ro