OpenHantek/OpenHantek6022 #291: Can't move/resize measurement box

wavelet2, 2022-04-17:
When I enable measurements the yellow/blue box appears in the center of the screen, 2 grids wide by 2 grids high.  I am unable to move or resize this box by clicking/dragging the cursor.  If I click and drag, the white vertical markers appear and follow the cursor movement instead.  Am I doing something wrong here?

I get the same behavior with multiple versions of openhantek, scope fw and operating systems.  Happy to provide details of a particular setup if needed but this seems to me like some sort of user error.  Just wondering if there's an obvious solution before I take things further.

Leonleon33, 2022-04-17:
Hello, I went through the same issue, if it can help the steps I followed
to solve it are
1) enable measurement
2) click "ON"
3) That is the thing I was missing : click on the channel you want ex "CH1"
it will be highlighted !
4) move or resize the box

wavelet2, 2022-04-17:
Thanks!  That is indeed the missing step.

closed 2022-04-17 by wavelet2