OpenHantek/OpenHantek6022 #296: MacOS build broken

Ho-Ro, 2022-04-30:
**Describe the bug**
The MacOs build stops during creation of the `*.dmg` package:

```
CPack: Create package
CPack: - package: packages/openhantek_20220429-d48201a-1_osx_x86_64.tar.gz generated.
ERROR: Could not parse otool output line: "\t/usr/local/opt/gcc/lib/gcc/11/libgcc_s.1.1.dylib (compatibility version 1.0.0, current version 1.1.0, reexport)"
ERROR: Could not parse otool output line: "\t/usr/local/opt/gcc/lib/gcc/11/libgcc_s.1.1.dylib (compatibility version 1.0.0, current version 1.1.0, reexport)"
Error: Process completed with exit code 1.
```

This error also occurs with a previous build that was successful 4 days ago. So it is not caused by any OH6022 changes. I suspect this is from a regression in the MacOs build system. Due to a lack of Mac on my end, I will not investigate further and leave it to a volunteer.

Ho-Ro, 2022-05-09:
As of 3.3.0, macOS binaries (bundles) will no longer be provided due to this issue unless a volunteer works out, provides and supports the bundle creation for GitHub Actions.

benjones, 2022-06-26:
I have a mac so I can try stuff out and maybe do some small fixes, but I'd need some direction to know where to look/what to do

Ho-Ro, 2022-06-27:
The 1st test you can do is to check [the mac build](https://github.com/dianlight/OpenHantek6022/suites/6965851183/artifacts/272042831) provided by @dianlight. I will pull his PR when I am back from vacation next weekend.

benjones, 2022-06-29:
I ran it with my 6022BE on an M1 mac and it seems to work fine, aside from it not being from an identified developer.

Ho-Ro, 2022-07-02:
Solved by #312

closed 2022-07-02 by Ho-Ro