OpenHantek/OpenHantek6022 #299: the program breaks down at startup (MAC OS)

pecherskiy-v, 2022-05-06:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenHantek [43496]
Path:                  /Users/USER/*/OpenHantek.app/Contents/MacOS/OpenHantek
Identifier:            OpenHantek
Version:               d48201a (20220506-d48201a)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-05-06 20:01:23.7296 +0300
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Anonymous UUID:        01531E3B-7F42-3B41-A099-A3258C0718AE

Sleep/Wake UUID:       ED8E347E-92BC-4034-82AF-D3EA2B771C5F

Time Awake Since Boot: 41000 seconds
Time Since Wake:       2270 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x00000001043a4000
Exception Codes:       0x0000000000000032, 0x00000001043a4000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 2 

VM Region Info: 0x1043a4000 is in 0x1043a4000-0x1043d0000;  bytes after start: 0  bytes before end: 180223
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      mapped file                 10434c000-1043a4000    [  352K] r--/rwx SM=COW  ...t_id=44c544f3
--->  mapped file                 1043a4000-1043d0000    [  176K] r-x/rwx SM=COW  ...t_id=44c544f3
      VM_ALLOCATE (reserved)      1043d0000-1043fc000    [  176K] rw-/rwx SM=NUL  ...(unallocated)

Thread 0 Crashed:
0   dyld                          	       0x104484014 dyld3::MachOFile::isMachO(Diagnostics&, unsigned long long) const + 20
1   dyld                          	       0x1044652e0 dyld4::Loader::mapSegments(Diagnostics&, dyld4::RuntimeState&, char const*, unsigned long long, dyld4::Loader::CodeSignatureInFile const&, bool, dyld3::Array<dyld4::Loader::Region> const&, bool, bool, dyld4::Loader::FileValidationInfo const&) + 1096
2   dyld                          	       0x1044652e0 dyld4::Loader::mapSegments(Diagnostics&, dyld4::RuntimeState&, char const*, unsigned long long, dyld4::Loader::CodeSignatureInFile const&, bool, dyld3::Array<dyld4::Loader::Region> const&, bool, bool, dyld4::Loader::FileValidationInfo const&) + 1096
3   dyld                          	       0x10446ab6c invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 68
4   dyld                          	       0x10446a50c dyld4::JustInTimeLoader::withRegions(dyld3::MachOAnalyzer const*, void (dyld3::Array<dyld4::Loader::Region> const&) block_pointer) + 292
5   dyld                          	       0x10446aac0 invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 480
6   dyld                          	       0x10446fd3c dyld4::SyscallDelegate::withReadOnlyMappedFile(Diagnostics&, char const*, bool, void (void const*, unsigned long, bool, dyld4::FileID const&, char const*) block_pointer) const + 132
7   dyld                          	       0x10446a8ac dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 204
8   dyld                          	       0x1044646e4 invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 1384
9   dyld                          	       0x104463bd4 dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 780
10  dyld                          	       0x104463800 invocation function for block in dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 148
11  dyld                          	       0x104459ec4 dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, dyld3::Platform, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 672
12  dyld                          	       0x104463754 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 172
13  dyld                          	       0x104463f74 dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 864
14  dyld                          	       0x104468b44 invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 380
15  dyld                          	       0x104485270 invocation function for block in dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 148
16  dyld                          	       0x104451f98 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 168
17  dyld                          	       0x1044850b8 dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 172
18  dyld                          	       0x1044688ac dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 164
19  dyld                          	       0x1044555bc dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 1092
20  dyld                          	       0x104455068 start + 484

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x00000001043a4000   x1: 0x000000016bd2a008   x2: 0x000000000002c000   x3: 0x0000000000040012
    x4: 0x0000000000000003   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x00000001044c862c   x9: 0x00000001044c9ea8  x10: 0x000000001e000000  x11: 0x0800000000034000
   x12: 0x0000000000000001  x13: 0x0000000000000001  x14: 0x0000000000021850  x15: 0x0000000000000000
   x16: 0x00000000000000c5  x17: 0x6ae100016bd297e8  x18: 0x0000000000000000  x19: 0x000000016bd2a008
   x20: 0x000000010424c060  x21: 0x0000000000058000  x22: 0x0000000000000003  x23: 0x000000016bd29af8
   x24: 0x0000000000000040  x25: 0x0000000000000000  x26: 0x000000016bd2987c  x27: 0x0000000000000000
[... 127 lines cut ...]
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "6216ae152a40e71046e16225"
      },
      "deploymentId" : 240000016
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,1, BootROM 7459.101.3, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Feb  8 2022 01:51:44 version 20.90.39.0.8.7.114 FWID 01-510a2aee
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

Ho-Ro, 2022-05-06:
Please provide more info:
- How did you create your mac version?
- Is it a regression?
- Which previous version did work?
- Did you check my pre-compiled binaries / packages?

pecherskiy-v, 2022-05-06:
your assembled version 3.2.5 has earned
older versions do not work, including those you have collected
I tried to build from the source code myself, even version 3.2.5 does not work, but it is assembled without errors

Ho-Ro, 2022-05-07:
> Code Type: ARM-64 (Native)

I saw that you are on an ARM system, while the pre-built packages on GitHub are [Intel](https://github.com/OpenHantek/OpenHantek6022/runs/6240131416?check_suite_focus=true#step:6:895).
Please give precise information about your build/target system and how you worked.

Latest 3.2.5 package for mac was downloaded 375 times and I got no complaint, so I suspect that this version worked for a lot of people.
```
OpenHantek/OpenHantek6022/releases/latest                 updated   dl
openhantek_20211216-31c5ee2-1_osx_x86_64.dmg 2021-12-16T15:05:08Z  375
```

Maybe a user can give positive feedback.

Ho-Ro, 2022-06-10:
```
OpenHantek/OpenHantek6022/releases/tags/3.2.5             updated   dl
openhantek_20211216-31c5ee2-1_osx_x86_64.dmg 2021-12-16T15:05:08Z  430
```
Even more downloads without another complaint after 1 month, I'll close for now. Please reopen if you can provide more information.

closed 2022-06-10 by Ho-Ro