OpenHantek/OpenHantek6022 #302: DSO-6022BE Not Opening

Tajnology, 2022-05-21:
OpenHantek cannot open 6022BE (Digitech QC-1929), unless the device is removed and re-inserted while OpenHantek is running.

Steps to reproduce the behavior:
1. Uninstall all drivers.
2. Install Hantek_6022BE_loader, *_openht and _sigrok drivers.
3. Re-insert USB.
4. Start OpenHantek.exe.
5. Select Device.

Expect the device to open correctly.

Demonstration:
[https://youtu.be/aBFS3D8WXr0](url)

Version 3.3.0-rc3

Tajnology, 2022-05-21:
`OpenHantek6022 (3.3.0-rc3)
1 ms: OpenHantek6022 - version 3.3.0-rc3
2 ms: create openHantekApplication
5 ms: load translations for locale "en_AU"
5 ms: init libusb
6 ms: show splash screen
 FindDevices::FindDevices()
  FindDevices::updateDeviceList()
  +++ "0x1a00000004b50210" "DSO-6022BE"
  DevicesListModel::updateDeviceList() "DSO-6022BE" cannot connect
  FindDevices::updateDeviceList()
  FindDevices::updateDeviceList()`

Ho-Ro, 2022-05-21:
Which was the last working OH6022 version? Did it work with 3.2.5?
@fgrieu you provided the `*.inf` files, do you have an idea how to solve?

Tajnology, 2022-05-22:
Sorry, did not mean to close that. I have tried now using the zadig drivers, and it works correctly with the current prerelease, 3.2.5 and 2.16. I will try using fgrieu's drivers on a different PC.

Tajnology, 2022-05-22:
Ok, I've had success. Here are the steps I took:
1. Insert USB
2. Uninstall device and remove drivers
3. Re-insert USB
4. Install Hantek_6022BE_loader.inf
5. Install Hantek_6022BE_openht.inf
6. Restart PC
7. Launch OpenHantek

I didn't reset my computer the first time I installed fgrieu's drivers, but I did when installing the zadig drivers. So, it seems a reset is necessary, contrary to fgrieu's Readme.

Tajnology, 2022-05-22:
By the way, Windows 11's Device Manager has an incredibly useful View > Devices by driver option, which allows you to uninstall drivers for devices that are not loaded.

closed 2022-06-10 by Ho-Ro

dzid26, 2022-12-20:
Some notes after using OpenHantek for a few months:

Recently I started having issues with my USB hub and Openhantek.
It is sometimes necessary to unplug and insert USB again while OpenHantek is searching for devices in order for the firmware to be replaced and show up as Hantek 6022BE openht in device manager.

Also, at the same time, the Openhantek configuration apparently got broken and the program would crash right after loading scopes. I solved it by removing settings in `Computer\HKEY_CURRENT_USER\Software\OpenHantek`.