OpenHantek/OpenHantek6022 #304: Size of viewpoint

nemanjan00, 2022-05-31:
**Describe the bug**

[image]

Viewpoint does not resize when I click with resize cursor and drag to resize

**To Reproduce**

Open program and try to resize

**Expected behavior**
Viewpoint should resize

**Screenshots**

[image]

**Computer environment (please complete the following information):**
 - OpenHantek version: b46ab64866477a37fb5478dcac5742a0d42263b3
 - OS: Linux
 - Distribution, version: Arch with Wayland
 - Video hardware:  Intel® UHD Graphics for 8th Generation Intel® Processors
 - OpenGL version: Mesa 22.1.0-1
 - Qt version: Qt 5.15.3
 - C++ compiler version: gcc 12.1.0

**Scope device (please complete the following information):**
 - Device Hantek 6022BE
 - Program top line OpenHantek6022 (3.3.0) - Device  DSO-6022-BE (FW0210)
 - Input signal: no
 - Probe setting:no
**Additional context**

```
OpenHantek6022 (3.3.0)
Graphic: 4.6 (Compatibility Profile) Mesa 22.1.0 - GLSL version 1.20
```

Ho-Ro, 2022-05-31:
If you use this *erroneous* laptop, please also check whether the sub-windows (e.g. the setup window) show comparable behaviour. But I suspect that [`DsoWidget`](https://github.com/OpenHantek/OpenHantek6022/blob/main/openhantek/src/dsowidget.cpp) (the black part of the scope) is causing this issue.

nemanjan00, 2022-06-12:
Hello,
There is similar behavior in other windows too

[image]

nemanjan00, 2022-06-12:
Ok, I figured it out. It is related to QT theme I used

closed 2022-06-12 by nemanjan00

nemanjan00, 2022-06-12:
Just in case you want to take a deeper look, since it is quite popular theme framework, it is Kvantum

Ho-Ro, 2022-06-13:
Thank you for this info, I can confirm this issue with Kvantum and Kvantum dark (never used this theme before). A simple workaround would be to select "Fusion" style and "Dark" theme in menu Oscilloscope/Settings/Colors.

Ho-Ro, 2022-06-14:
There is now a "hack" (c0a7fac) that switches to the dark fusion style when it detects "kvantum*" as the system style - you may have to manually correct the disturbed look once.

jvandenbroek, 2022-08-05:
Thanks, I had the same issue. Not sure about that hack @Ho-Ro , I've installed the git version from AUR and still had this issue with Kvantum theme.

Ho-Ro, 2022-08-05:
Which version does the program show in the top line?

jvandenbroek, 2022-08-25:
Sorry for my late response, but you are absolutely right. Didn't realize the git version of AUR was that outdated. Now I've installed the recently released regular version (also from AUR). The issue still persists with kvantum-dark, but setting Fusion style in settings solves this.

Ho-Ro, 2022-08-25:
> The issue still persists with kvantum-dark, but setting Fusion style in settings solves this.

Please unset Fusion, restart the program with `OpenHantek --verbose 1` and report the startup messages, normally it should select "Fusion" automatically if it detects "kvantum*". But nevertheless it's good that you found a workaround.