OpenHantek/OpenHantek6022 #309: Some errors when starting without an existing Version

a-n-d-i, 2022-06-04:
I downloaded Openhantek6022 for the first time today and it showed some weird behavior:

- no roll mode
- refresh button not showing at all
- start/stop in single trigger mode does not clear the screen
- possibly more

[image]

I installed a random version from last year, worked fine. Reinstalled 3.3.0, worked fine. Deleted the folder in ~.config, broken. 

Might be related to #308

Ho-Ro, 2022-06-04:
> no roll mode

The roll mode only makes sense with a slow time base > 100 ms/div and is therefore only activated then. If you use an old configuration with activated roll mode at a faster time base, the roll mode remains, but switches to the slow time base of 200 ms/div.

> refresh button not showing at all

The refresh button is only useful in roll mode and is only displayed in this mode.

> start/stop in single trigger mode does not clear the screen

A free running screen is displayed until a new triggered trace is available. 

> Deleted the folder in ~.config, broken.

I did the same, removed ~/.config/OpenHantek, still works as I expect.

Ho-Ro, 2022-06-14:
@a-n-d-i Any new findings?

a-n-d-i, 2022-06-27:
@Ho-Ro No, haven't used the software any more, bought the scope to keep on my boat in case something to troubleshoot comes up.

Ho-Ro, 2022-07-02:
Ok, then I will close for now - please reopen if necessary.

closed 2022-07-02 by Ho-Ro