OpenHantek/OpenHantek6022 #314: Trying to open pre-build openhantek_20220703-e5f0e4aon macOS 11.6.6 gives "This App requires macOS 12.0 or newer"

warpme, 2022-07-06:
**Describe the bug**
Trying to open https://github.com/OpenHantek/OpenHantek6022/releases/download/3.3.0.1.macos/openhantek_20220703-e5f0e4a-1_osx_x86_64.dmg on macOS 11.6.6 gives error dialog "This App requires macOS 12.0 or newer"

**Computer environment (please complete the following information):**
 - OpenHantek version: [openhantek_20220703-e5f0e4a-1_osx_x86_64.dmg]
 - OS: [macOS 11.6.6]
 - Distribution, version: [pre-build macOS .dmg]
 - Video hardware: [rMBP Mid2014 with Intel Iris]

I this expected?

Ho-Ro, 2022-07-07:
Long answer: The macOS build was broken some time ago (#296) due to a regression caused by an update of the macOS build system (GitHub actions). @dianlight found a *"quick and dirty "* solution (#312), but it requires macOS 12 and only supports the Intel target. 
Due to a missing Mac on my end, I won't investigate further and will leave it to a volunteer to get the package build completely working again.

warpme, 2022-07-30:
FYI: i got compiling and well working current master code on my macOS 11.6.8 + Xcode 

hack to get it packaging on macOS Big Sur is:

```
mkdir -p /usr/local/opt/qt5/lib/libgcc_s.1.1.dylib.framework
ln -sf /usr/local/opt/gcc/lib/gcc/11/libgcc_s.1.1.dylib /usr/local/opt/qt5/lib/libgcc_s.1.1.dylib.framework/libgcc_s.1.1.dylib
``` 

to automatise periodic builds i'm using script like this:
 

```
echo "--> Symlinking GCC11 libgcc_s.1.1.dylib"

mkdir -p /usr/local/opt/qt5/lib/libgcc_s.1.1.dylib.framework
ln -sf /usr/local/opt/gcc/lib/gcc/11/libgcc_s.1.1.dylib /usr/local/opt/qt5/lib/libgcc_s.1.1.dylib.framework/libgcc_s.1.1.dylib

echo "--> Pulling git"

git pull
git submodule update --init --recursive

echo "--> building"

export LDFLAGS="-L/usr/local/opt/qt@5/lib"
export CPPFLAGS="-I/usr/local/opt/qt@5/include"
export PATH="/usr/local/opt/qt@5/bin:$PATH"

mkdir -p build
rm -rf build/*
cd build
cmake ..
#
make -j8
#
# now the target was created in subdir openhantek
# .. either as single binary OpenHantek, then you're done
# .. or as a bundle if enabled in ../../openhantek/CMakeLists.txt
# .. but this bundle is still a template as the dynlibs are not yet bundled
# .. this magic will happen now
#

echo "--> packaging"

cd openhantek
#
# deploy all necessary Qt dynlibs into the bundle
macdeployqt OpenHantek.app -always-overwrite -verbose=2
#
# find all other dependencies, and their dependencies, and their... (you got it!)
python ../../utils/macdeployqtfix/macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek $(brew --prefix qt5)
#

echo "--> creantng DMG"

# finally create OpenHantek.dmg from OpenHantek.app
create-dmg --volname OpenHantek --volicon ../../openhantek/res/images/openhantek.icns --window-pos 200 120 \
  --window-size 800 400 --icon-size 100 --icon "OpenHantek.app" 200 190 --skip-jenkins \
  --hide-extension "OpenHantek.app" --app-drop-link 600 185 --eula ../../LICENSE OpenHantek.dmg OpenHantek.app

```

Ho-Ro, 2022-07-30:
@warpme please check my latest unstable that was created on macos-11 using your hack from above:
https://github.com/OpenHantek/OpenHantek6022/releases/download/unstable/openhantek_20220730-906d4ab-1_osx_x86_64.dmg

warpme, 2022-07-30:
@Ho-Ro,
Your binary works perfectly on my rMBP :-)

Ho-Ro, 2022-07-31:
@warpme I have just released stable version 3.3.1 with your solution. Thank you for your support.

closed 2022-07-31 by Ho-Ro

dhysuiej, 2023-05-23:
I'm getting an error saying You have macOS 10.15.7. The application requires macOS 11.7 or later.

Is this because of some missing feature on macOS 10.15.7 or could it be built without any problems for that system just modifying the target macOS version?