OpenHantek/OpenHantek6022 #316: OpenHantek6022 on Debian (fresh install, latest version) requires multi launch, shows FW209; crashes when I touch the probe

rechnermusik, 2022-07-14:
Running a Debian VM on MacOS. When I plug in my 6022BE, the light is off. I need to launch OpenHantek6022 a couple times to get it to flash red, then eventually green.

Once the software opens, it shows FW209 in the title bar. Software seems stable until I even pick up the probe with my hand.

Does this seem like a software issue, hardware issue, or both?

Please let me know how I can provide more detailed information for troubleshooting. Thank you!

https://user-images.githubusercontent.com/50645813/178927149-4289a3a8-bcbf-49a8-aec9-fdb3e480c0e0.mov

Ho-Ro, 2022-07-14:
Strange, this commit 1033 should show FW 0210 (the FW is intrinsic part of the binary). How did you install?
[image]

Ho-Ro, 2022-07-15:
Very very strange, I checked thoroughly the binaries from GitHub and the Debian as well as the Windows package contain FW0210:
```
strings OpenHantek | grep 40B5042260
:203D000012010002FFFFFF40B50422601002010203010A0600020000004001000902890016
strings OpenHantek.exe | grep 40B5042260
:203D000012010002FFFFFF40B50422601002010203010A0600020000004001000902890016
```
where :203D000012010002FFFFFF40**B50422601002**010203010A0600020000004001000902890016 is the USB idVendor 04B5, the USB idProduct 6022 and the USB bcdDevice 0210 of the FW (all values are in little-endian format)

Also in #315 you can see a [screen shot](https://user-images.githubusercontent.com/36264810/178125597-fa5bf0b1-1e4f-4394-b5ab-ff7317ecd0af.png) with FW0210 and also the YT videos show this FW:
[image]

**Where does this old version come?**

- When started the OpenHantek binary searches the USB busses for compatible HW.
- If it detects a freshly connected device (without FW, VID=04B4, PID=0622) then it uploads the FW  that is intrinsic part of the binary. 
- If it detects a device with previously uploaded FW (VID=04B5, PID=0622) then it checks the FW version against the requirement in the binary. If the FW is too old it will be replaced. 

I found an [issue](https://github.com/OpenHantek/OpenHantek6022/blob/e5f0e4a2b25f9738cd66afe90a1ada6440ce51e5/openhantek/src/hantekdso/models/modelDSO6022.cpp#L117-L124) in the program that accepts also FW 0208 and 0209. But this doesn't clarify where the old version in the device comes.

Issue fixed in latest unstable, please check:
If you start the binary from a terminal window as `OpenHantek --verbose 3` you will get some debug info, please report the 1st part 'til 'use device "DSO-6022BE" serial number ...':
```
OpenHantek6022 (20220715 - commit 1034)
1 ms: OpenHantek6022 - version 20220715 - commit 1034
1 ms: create openHantekApplication
103 ms: set "Fusion" style
104 ms: set RT FIFO scheduler
104 ms: load translations for locale "de_DE"
104 ms: init libusb
112 ms: show splash screen
 FindDevices::FindDevices()
  FindDevices::updateDeviceList()
  +++ "0x1100000004b40000" "DSO-6022BE"
  DevicesListModel::updateDeviceList() "DSO-6022BE" upload firmware
  FindDevices::updateDeviceList()
  --- "0x1100000004b40000" "DSO-6022BE"
  FindDevices::updateDeviceList()
  +++ "0x1100000004b50210" "DSO-6022BE"
  DevicesListModel::updateDeviceList() "DSO-6022BE" can connect
2036 ms: use device "DSO-6022BE" serial number "C3B36055284A"
``` 
1. It sees device w/o FW (VID 04b4, FW 0000) added to USB
```
+++ "0x1100000004b40000" "DSO-6022BE"
```
2. Uploads FW, the device w/o FW vanishes from USB
```
--- "0x1100000004b40000" "DSO-6022BE"
```
3. Finally a device with FW0210 (VID 04b5, FW 0210) appears on USB
```
+++ "0x1100000004b50210" "DSO-6022BE"
```

rechnermusik, 2022-07-21:
Strange, today it shows 0210 firmware using the stable build. But it still crashes as soon as I plug the probe into the BNC jack.

[image]

rechnermusik, 2022-07-21:
Actually, I don't even need to plug the probe, it will crash on it's own.
[image]

Ho-Ro, 2022-07-21:
It looks like the USB communication stops and as a result the program crashes, same behaviour when I unplug my scope (bulkReadMulti error). What does `dmesg` say?

rechnermusik, 2022-07-21:
[image]

rechnermusik, 2022-07-21:
[image]

Ho-Ro, 2022-07-22:
This is my `dmesg` output when I plug in the scope and start the program:
```
[Jul22 09:51] usb 1-1: new high-speed USB device number 24 using ehci-pci
[  +0,156445] usb 1-1: New USB device found, idVendor=04b4, idProduct=6022, bcdDevice= 0.00
[  +0,000009] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ +15,956007] usb 1-1: USB disconnect, device number 24
[  +1,531516] usb 1-1: new high-speed USB device number 25 using ehci-pci
[  +0,165112] usb 1-1: New USB device found, idVendor=04b5, idProduct=6022, bcdDevice= 2.10
[  +0,000008] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000006] usb 1-1: Product: DSO-6022BE
[  +0,000004] usb 1-1: Manufacturer: OpenHantek
[  +0,000004] usb 1-1: SerialNumber: C3B36055284A

```
Other devices on the same USB bus? I find my 6022 as device 25 on Bus 001 and the tree shows that there's only this device 25 on Bus01.Port1.Port1
```
lsusb
...
Bus 001 Device 025: ID 04b5:6022 ROHM LSI Systems USA, LLC DSO-6022BE
...
lsusb -t
...
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/6p, 480M
    |__ Port 1: Dev 25, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
...
```
Is this a native or virtual system, on my (old) ThinkPad virtual USB is not stable.

Ho-Ro, 2022-08-27:
It seems as if the problem has magically solved itself - I am closing this issue.

HTD, 2022-10-07:
I'm not sure about it. I had the same issue on the latest unstable Windows version. The problem was triggered not just with touching the probe but feeding the scope with a nice digital signal. I just set the scope to reproduce the issue, then CLOSED the app to have my settings saved and this caused the problem doesn't occur anymore. Like it was something in the settings that is fixed when the app is normally closed, not crashed.

Ho-Ro, 2022-10-07:
Just an idea, if this crash happens it could solve the issue when you start the next time with `OpenHantek --resetSettings`.
I can check if there's a regression introduced by a change in config storage structure - in this case I can increase the config version number - this will automatically restart with a clean config on first invocation.

rechnermusik, 2022-10-07:
Hi everyone, I've been away hospitalised for a while but I came back to test with the latest version and Ho-Ro's suggestions. Still getting similar results. Crashes immediately when the probe is already connected, or crashes at the point when I (barely) touch the test/ground terminals with my hand, or begin to plug the probe into the BNC connector.

[image]

osboxes@osboxes:~$ OpenHantek --resetSettings
OpenHantek6022 (3.3.1)
Graphic: 2.1 Mesa 20.3.5 - GLSL version 1.20
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
Sending control command e6 failed: No such device (it may have been disconnected)
[... cut ...]
Graphic: 2.1 Mesa 20.3.5 - GLSL version 1.20
bulkReadMulti: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "No such device (it may have been disconnected)"
OpenHantek6022 controlWrite: stop sampling failed:  "No such device (it may have been disconnected)"
controlWrite: Getting sample data failed:  "No such device (it may have been disconnected)"
controlWrite: stop sampling failed:  "No such device (it may have been disconnected)"
has stopped after 19 s
osboxes@osboxes:~$

Ho-Ro, 2022-10-08:
If this issue would be triggered by an incompatible setup storage it would ge away when calling with `OpenHantek --resetSettings`, but also without the option OH would announce a kind of "index out of range" or similar on the command line before crash. It seems your issue is caused by a weak USB connection - did you check the cable, use another port, check on another computer etc.

commit 02ba1617 (2022-10-18): README and doc update (building); get FW version from dso602x_fw_version.h

increase CONFIG_VERSION, fix #316 (second part)
closed 2022-10-18 by Ho-Ro