OpenHantek/OpenHantek6022 #334: Readme needs an update

munching, 2022-11-17:
Hi,

"USB Access" section, "Windows" department seems to contain some outdated info. There is no "driver" folder in the binary distribution neither is there a file named "OpenHantek.inf"

Also it would be very nice to have some additional info for complete beginners like me: what exactly does loader, openht and sigrok do? Not how to install them but rather why to install them? It's unclear, do I need loader, loader + one of the two or all three. 

Another concerning point is that it was mentioned that at some point something is written to scope's firmware. Does this mean if I install the OpenHantek drivers and use the scope it will render the scope unusable with original software (on another PC with original driver & software)?

Thanks

closed 2022-11-17 by Ho-Ro

munching, 2022-11-17:
Hi @Ho-Ro 
Thank you for the explanations. I didn't realize the readme refers to the unstable, was looking at the latest stable build. I also have another issue to report regarding the UI but as there was mentioned, windows-related issues are not accepted :( This is easily replicated in demo mode using specific screen resolution.

Ho-Ro, 2022-11-17:
Just open a new issue, but I cannot promise support because I do not use Windows and have no Windows PC to test. But if this is a general issue also other users should be able to reproduce and debug.
As I said: _No support for non-Linux related issues unless a volunteer steps in!_