OpenHantek/OpenHantek6022 #335: Settings window collapses under some conditions

munching, 2022-11-18:
In windows distribution Oscilloscope -> Settings window can't be resized vertically. This is only encountered if Display -> "Scale and layout" is set to 150% (for 1920x1080 resolution) in display settings of windows. Under such conditions once the oscilloscope settings window is shown, it can't be fit into screen and is cut at the bottom so some parts can't be seen. If you try to resize the window vertically, it becomes hidden but somehow stays active as a modal window, thus preventing any other actions with the OpenHantek6022 (this buggy settings window can still be closed by pressing Escape button). This can be encountered in both normal and demo mode.

The issue is currently minor, I checked it on other scale modes and there aren't any parameters actually not accessible in the settings window. But perhaps it would be useful to be aware of such issue in case more settings are added to the bottom of the window in the future.

Ho-Ro, 2022-11-18:
> Display -> "Scale and layout" is set to 150% (for 1920x1080 resolution)  ...  Under such conditions once the oscilloscope settings window is shown, it can't be fit into screen and is cut at the bottom so some parts can't be seen. 

150% at 1920x1080 gives a huge window. It could help to set the font size to something smaller than the default. To achieve this without access to the complete visible settings window just use the command line option and start the program with e.g. `OpenHantek -s 6`. After stopping the program this value is saved permanently. Also disabling Fusion style gives some percent smaller window size.
As a kind of fix I also rearranged the settings windows to decrease the height a little bit - please check the upcoming unstable.

Ho-Ro, 2022-12-13:
No more complaints - closed.

closed 2022-12-13 by Ho-Ro