OpenHantek/OpenHantek6022 #339: crash on device disconnect

hramrach, 2022-12-17:
**Describe the bug**
crash when device is disconnected.

**To Reproduce**
Run OpenHantek, plug in device, wait for data display, unplug device.

**Expected behavior**
No crash - clean shutdown or wait for device reconnect.

**Screenshots**
```
OpenHantek6022 (3.3.1)
Graphic: 4.2 (Compatibility Profile) Mesa 22.2.4 - GLSL version 1.20
bulkReadMulti: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "Input/Output Error"
controlWrite: Getting sample data failed:  "No such device (it may have been disconnected)"
OpenHantek6022 controlWrite: stop sampling failed:  "No such device (it may have been disconnected)"
controlWrite: stop sampling failed:  "No such device (it may have been disconnected)"
has stopped after 2 s
malloc(): unaligned tcache chunk detected
Aborted (core dumped)
```

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.3.2
 - OS: Linux
 - Distribution, version: openSUSE Tumbleweed (amd64)
 - Video hardware: Mesa Intel(R) HD Graphics 2500 (IVB GT1)
 - OpenGL version: 4.2 (Core Profile) Mesa 22.2.4
 - Qt version: libQt5Core5-5.15.7+kde167-2.1.x86_64
 - C++ compiler version: gcc (SUSE Linux) 12.2.1 20221020

**Scope device (please complete the following information):**
 - Device Hantek 6022BL
 - Program top line OpenHantek6022 (3.3.1) - Device DSO-6022BL (FW0210)
 - Input signal: any
 - Probe setting: any

Ho-Ro, 2022-12-17:
WONTFIX
Please check #337

closed 2022-12-17 by Ho-Ro

hramrach, 2022-12-17:
It potentially uses up disk space when it crashes rather than exiting cleanly.

Ho-Ro, 2022-12-17:
Please elaborate.

hramrach, 2022-12-17:
`Aborted (core dumped)`

That means the core is stored somewhere.

Ho-Ro, 2022-12-17:
Ok, understood. Debian does not dump the core, so there should also be a solution for openSUSE.

hramrach, 2022-12-17:
The universal solution is for the program to not crash in the first place.

Ho-Ro, 2022-12-17:
The best solution would be to keep  the USB connection untouched. If you cannot ensure this, either do not use the program or modify it according to your needs.
EOD