OpenHantek/OpenHantek6022 #341: Compilation errors under Fedora 37

spam-receiver, 2023-01-04:
Errors like this:
_mainwindow.cpp:134:61: error: «VERSION» was not declared in this scope;
main.cpp:94:46: error: «VERSION» was not declared in this scope;_

In _main.cpp_ I commented out check of "VERSION" definition:
#ifndef VERSION
//#error "You need to run the cmake buildsystem!"
#endif
#include "OH_VERSION.h"

Then changed "VERSION" to "OH_VERSION" in _main.cpp_ and _mainwindow.cpp_

After that program was compiled successfully.

And I added local docs into _openhantek.spec_:

%doc README.md CHANGELOG docs/OpenHantek6022_User_Manual.pdf docs/HANTEK6022_AC_Modification.pdf docs/HANTEK6022_Frequency_Generator_Modification.pdf CODE_OF_CONDUCT

[openhantek.spec.zip](https://github.com/OpenHantek/OpenHantek6022/files/10340746/openhantek.spec.zip)

Vascom, 2023-01-04:
I can confitm that problem.
But need some answer or may be patch from upstream.

spam-receiver, 2023-01-04:
> I can confitm that problem. But need some answer or may be patch from upstream.

[openhantek-3.3.2-1.fc37.src.rpm.txt](https://github.com/OpenHantek/OpenHantek6022/files/10345986/openhantek-3.3.2-1.fc37.src.rpm.txt)

Ho-Ro, 2023-01-06:
> mainwindow.cpp:134:61: error: «VERSION» was not declared in this scope;
main.cpp:94:46: error: «VERSION» was not declared in this scope;

How do you build the binary?

`VERSION` is defined in the `Makefile` created by CMake in `CMakeList.txt`:

```cmake
if (GIT_EXECUTABLE AND EXISTS "${CMAKE_SOURCE_DIR}/.git")
    execute_process(
        COMMAND ${GIT_EXECUTABLE} describe --tags --dirty
        WORKING_DIRECTORY ${CMAKE_SOURCE_DIR}
        OUTPUT_VARIABLE VERSION
        OUTPUT_STRIP_TRAILING_WHITESPACE
    )
message( STATUS "VERSION: ${VERSION}" )
endif()

# ...

# enable extra feature(s)
if( ${CMAKE_VERSION} VERSION_LESS "3.12.0" ) # deprecated, but still used in older Ubuntu releases
    # Enable C++ standard library hardening -> cheap range checks for C++ arrays, vectors, and strings.
    add_definitions( -D_GLIBCXX_ASSERTIONS )
    add_definitions( -D_USE_MATH_DEFINES )
    if ( DEFINED VERSION )
        add_definitions( -DVERSION="${VERSION}" )
    endif()
else() # 'add_compile_definitions()' was introduced with CMake 3.12
    add_compile_definitions( _GLIBCXX_ASSERTIONS )
    add_compile_definitions( _USE_MATH_DEFINES )
    if ( DEFINED VERSION )
        add_compile_definitions( VERSION="${VERSION}" )
    endif()
endif()
```

What is the output of the call to `cmake ..` in directory `build`?

Do not use `OH_VERSION` from `OH_VERSION.h`, the comments in this file are outdated.

`VERSION` is created as the output of `git describe --tags --dirty` and should look like `3.3.2-nn-gxxxxxxx` where `nn` is the number of commits after the tagged version (e.g. 3.3.2) and `xxxxxxx` is the git commit hash.

If you want to define your own versioning, do not rely on `OH_VERSION` as this variable is only valid if the commit is tagged and then renamed to `LAST_OH_VERSION` if the commit is not tagged.

spam-receiver, 2023-01-06:
> How do you build the binary?

As always:

_rpmbuild -ba openhantek.spec_

spam-receiver, 2023-01-06:
[LinuxBuild_output.txt](https://github.com/OpenHantek/OpenHantek6022/files/10363598/LinuxBuild_output.txt)

Ho-Ro, 2023-01-07:
Your setup is wrong, you're not building out of a git repo!
CMake fails to create the define for `VERSION`:
```
[spam-receiver@lab OpenHantek6022-3.3.2]$ ./LinuxBuild
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib64/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.39.0")
-- GIT_COMMIT_HASH:
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

Please apply this patch and try e.g. `VERSION=3.3.2-fc37 cmake ..`
The modification takes an env variable to overrule the intrinsic version definition.

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index b8e0738..2e348f6 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -17,15 +17,21 @@ INCLUDE_DIRECTORIES(".")
 
 find_package(Git)
 
-if (GIT_EXECUTABLE AND EXISTS "${CMAKE_SOURCE_DIR}/.git")
-    execute_process(
-        COMMAND ${GIT_EXECUTABLE} describe --tags --dirty
-        WORKING_DIRECTORY ${CMAKE_SOURCE_DIR}
-        OUTPUT_VARIABLE VERSION
-        OUTPUT_STRIP_TRAILING_WHITESPACE
-    )
-message( STATUS "VERSION: ${VERSION}" )
+if(DEFINED ENV{VERSION})
+    set(VERSION $ENV{VERSION})
+else()
+    if (GIT_EXECUTABLE AND EXISTS "${CMAKE_SOURCE_DIR}/.git")
+        execute_process(
+            COMMAND ${GIT_EXECUTABLE} describe --tags --dirty
+            WORKING_DIRECTORY ${CMAKE_SOURCE_DIR}
+            OUTPUT_VARIABLE VERSION
+            OUTPUT_STRIP_TRAILING_WHITESPACE
+        )
+    else()
+        set(VERSION "UNKNOWN")
+    endif()
 endif()
+message(STATUS "VERSION: ${VERSION}")
 
 # Use CPack to make tgz/deb/rpm/zip installer packages
 include(cmake/CPackInfos.cmake)
@@ -48,7 +54,7 @@ if( ${CMAKE_VERSION} VERSION_LESS "3.12.0" ) # deprecated, but still used in old
     add_definitions( -D_GLIBCXX_ASSERTIONS )
     add_definitions( -D_USE_MATH_DEFINES )
     if ( DEFINED VERSION )
-        add_definitions( -VERSION="${VERSION}" )
+        add_definitions( -DVERSION="${VERSION}" )
     endif()
 else() # 'add_compile_definitions()' was introduced with CMake 3.12
     add_compile_definitions( _GLIBCXX_ASSERTIONS )
```

spam-receiver, 2023-01-07:
Building package by _rpmbuild_ became successfull with the help of addition _export VERSION=%{version}_ to section _%build_ in _openhantek.spec_. Thank you for patching _CMakeLists.txt_!

[openhantek.spec](https://github.com/OpenHantek/OpenHantek6022/files/10367318/openhantek.spec.zip)

[CMakeLists.txt](https://github.com/OpenHantek/OpenHantek6022/files/10367314/CMakeLists.txt)

Awaiting Vasily's comment.

Vascom, 2023-01-08:
Patch is good.

Vascom, 2023-01-08:
But why VERSION retrivied only from git?
If I download release source archive I must set version manually?

Ho-Ro, 2023-01-08:
OH doesn't follow the idea of few releases, but has the _rolling_ approach and a x.y.z release is more like a lighthouse than a milestone. Typically most _unstable_ releases can be used for daily work. So the correct way of using OH for a developer is to clone it on your PC, pull new commits, build a binary and use it. For me it is important that every version in the wild can be identified uniquely to react on issue reports and to identify regressions. The versioning with `git describe -dirty` ensures this unique id, it shows also if the user added own changes.
If you provide modified RPMs you can now define your own version that follows the fedora style.

Vascom, 2023-01-08:
OH at Fedora has no modifications in sources. Or you mean that each build with different build flags is modification?

Ho-Ro, 2023-01-08:
No, but every change to the source code is a change. Also changes to the CMake files or the documentation. I got error  descriptions that I couldn't see on my build and it was caused by some unknown modifications by the user.

SuperTux88, 2023-01-08:
I find it very unexpected, that you need to set the version from external when using the official unmodified source-code assets from the [release page](https://github.com/OpenHantek/OpenHantek6022/releases/tag/3.3.2), especially since none of the build instructions tell you that you need to manually set the version. Also why do you need to set the version (to some possible arbitrary or wrong value) if the correct version is in the archive as `OH_VERSION`, why is that there if it's not used for anything?

The git version makes sense when building unreleased versions directly from git (so you know which commit it was), but when building an unmodified released version, wouldn't it be better to use the correct version that's in the release and you know it's correct (if you want people to report issues with the correct version)? I assume previous releases used exactly this version information, so why was that removed?

closed 2023-01-08 by Ho-Ro

Ho-Ro, 2023-01-08:
@SuperTux88 Thank you for your rant :)
>  I assume previous releases used exactly this version information, so why was that removed?

Using a tgz or zip encourages static behaviour where the user can change the source code without updating the version. 
The manual setting was a quick hack to make the fedora guys happy. 
The [build instructions](https://github.com/OpenHantek/OpenHantek6022#building-openhantek6022-from-source) tell that you should use `git clone ...` to get the source code when you want to build your own version. I didn't think of anyone using zip / tgz when you can have git, but I will restore the precedence of OH_VERSION over VERSION to avoid further annoying discussions.

Vascom, 2023-01-08:
Thanks.
Really we can use git hash as version if upstream has no ordinary versioning. But it is not preferred way.

So I am glad to see often version releases instead of rare "stable" versions.

SuperTux88, 2023-01-08:
> Thank you for your rant :)

Wasn't really meant as rant, more as feedback that building from released archives is also a usecase. Sorry if that came across wrong. :)

> The [build instructions](https://github.com/OpenHantek/OpenHantek6022#building-openhantek6022-from-source) tell that you should use `git clone ...` to get the source code when you want to build your own version.

It says that using git is "The easiest way to get an up-to-date working code base" (not the only way), which I agree that if I just want to build the latest development state for myself, I would also just use git. But when building released versions for a package manager, I usually use the tgz of the release.

That said, thanks for the quick fix and updated release. :heart:

commit de9d2012 (2023-01-08): fix #340, fix #341, upversion to 3.3.2.1

these changes allow to build from tgz / zip source without git support