OpenHantek/OpenHantek6022 #350: ChromeOS crostini with  .deb installation; blank window BUT screenshot from program OK

omminp, 2023-02-24:
When program runs, window opens as expected and scope red light flashes.  Demo mode button loads a black window. Moving the cursor over the window shows popups , can access top menu pulldowns with the cursor. Restart the program and a black window appears presumably the 'scope run window.
Tried setting Fusion, no change. 
Using Export pulldown, saved screenshot and it is normal (attached).  Screenshot is with Cal voltage connected so USB appears to be working OK.  
Also installed the flatpak version and it has the same behaviour.  Looks like the problem is coming from OpenHantek6022.
I don't know enough about the workings of the ChromeOS crostini display to be of use - any help with this would be appreciated, the program looks awesome.  Thaks
[image]

Ho-Ro, 2023-02-24:
- You could start the program from a terminal and look for error messages.
- Then you can try different options, you'll get the info if you start the Program with `OpenHantek -h`.
- Increase the verbosity with `--verbose 1` or `--verbose 2` -  3, 4, 5 give even more output.
- Maybe the `--useXXX` options can help.
```
Usage: OpenHantek [options]

Options:
  -?, -h, --help           Displays help on commandline options.
  --help-all               Displays help including Qt specific options.
  -v, --version            Displays version information.
  -c, --config <File>      Load config file
  -d, --demoMode           Demo mode without scope HW
  -e, --useGLES            Use OpenGL ES instead of OpenGL
  --useGLSL120             Force OpenGL SL version 1.20
  --useGLSL150             Force OpenGL SL version 1.50
  -i, --international      Show the international interface, do not translate
  -f, --font <Font>        Define the system font
  -s, --size <Size>        Set the font size (default = 10, 0: automatic from
                           dpi)
  --condensed <Condensed>  Set the font condensed value (default = 87)
  --resetSettings          Reset persistent settings, start with default
  --verbose <Level>        Verbose tracing of program startup, ui and
                           processing steps
```

omminp, 2023-02-25:
Thanks for your help.
I can't see any error messages.  No useXXX options make any difference. It
seems likely a graphics issue, I have other linux apps working OK with
graphical output,  - GNU Image, Arduino IDE, Audacity, etc

***@***.***:~$ OpenHantek  --verbose 1
OpenHantek6022 (3.3.0.1)
0 ms: OpenHantek6022 - version 3.3.0.1
0 ms: create openHantekApplication
81 ms: set RT FIFO scheduler
81 ms: load translations for locale "en_US"
81 ms: init libusb
83 ms: show splash screen
1119 ms: use device "DSO-6022BE" serial number "4B812809DC63"
1119 ms: create DSO control thread
1136 ms: create settings object
1146 ms: create exporters
1146 ms: create post processing objects
1147 ms: setup OpenGL
1191 ms: set QFont(Arial,10,-1,5,50,0,0,0,0,0)
1191 ms: create main window
GLScope::initializeGL()
Graphic: 3.1 Mesa 21.2.6 - GLSL version 1.20
1424 ms: start DSO control thread
1424 ms: execute GUI main loop

until window is closed, when this line

QOpenGLVertexArrayObject::destroy() failed to restore current context

I am happy to try any suggestions - I appreciate you probably don't have a
chromeOS machine.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
25/02/23,
19:04:37

lschweiss, 2023-05-01:
Unfortunately, this problem persists on ChromeOS.   

It would be great to get this working on Chromebooks, which are more portable than most laptops.

Ho-Ro, 2023-05-01:
There will be no progress unless a volunteer with a ChromeOS system works on this issue - I do not have access to this kind of machine.

omminp, 2023-05-02:
Confirming that the issue persists with the most recent version of
OpenHantek6022.

Recent posts made me think again, there is a project called ChromeOS FLEX
that allows a free download to make a bootable USB stick that runs ChromeOS
on most hardware that supports Windows or MacOS.  It is identical to a
Chromebook ChromeOS except that it doesn't support android apps.

I installed ChromeOS FLEX on a low-end laptop that had trouble running
Windows, and it boots fine, and provided the laptop supports virtualisation
it will run the Linux container. I installed the most recent OpenHantek6022
and it has the same issue as reported previously.

If anybody was willing to volunteer to look at this issue, I am happy to
assist in making the ChromeOS FLEX install as pain free as possible, to the
stage of getting  OpenHantek6022 running in the container.

I appreciate this may not be a priority for the project, but if there is a
way forward by working around the need for specific hardware, I would be
ready to help.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
02/05/23,
17:20:42

dsoltisII, 2024-03-13:
I was able to get this to work on ChromeOS Flex by setting the following environment variable
LIBGL_ALWAYS_SOFTWARE=1 OpenHantek

Ho-Ro, 2024-03-13:
@dsoltisII Thank you for reporting!
@omminp Did this solution also solve your problem?

omminp, 2024-03-13:
Thanks very much!  It works perfectly for me.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
14/03/24,
10:01:24

Ho-Ro, 2024-03-14:
Good that this long runner is finally closed now.

closed 2024-03-14 by Ho-Ro

omminp, 2024-03-14:
yes it is.  I'm grateful for your perseverance - much appreciated. Hope
someone else can benefit as well

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
14/03/24,
21:41:19

Ho-Ro, 2024-03-14:
I will add this info to the documentation - let's see where it fits best.