OpenHantek/OpenHantek6022 #358: Building on Apple M1 silicon

varna9000, 2023-03-19:
Hi, just wante to share that I got a successful (although somewhat unstable) build on my Macbook Pro  with M1 (arm64). Here are the slightly modified steps:

```
cd ~/Downloads/OpenHantek6022
git submodule update --init --recursive
rm -rf build/
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH=$(brew --prefix qt5)
make -j4

cd openhantek

$(brew --prefix qt5)/bin/macdeployqt OpenHantek.app -always-overwrite -verbose=2

python3 ../../utils/macdeployqtfix/macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek $(brew --prefix qt5)
```

Once qt5 is installed with Homebrew, I had to :
1. Pass DCMAKE_PREFIX_PATH to cmake so it finds qt
2. Pass the correct path to macdeployqt 
3. Instead of python2, we should pass python3

Also, the default [font-size 10pt ](https://github.com/OpenHantek/OpenHantek6022/blob/main/openhantek/src/viewsettings.h#L15)is very small on Mac, so I had to increase it to 14 (maybe 16 would be even better)

The build is not very stable, but it works. This is the warning in the terminal:

```
qt.qpa.fonts: Populating font family aliases took 77 ms. Replace uses of missing font family "Arial" with one that exists to avoid this cost.
Graphic: 2.1 Metal - 83 - GLSL version 1.50
QOpenGLShader::compile(Vertex): ERROR: 0:2: '' :  version '150' is not supported

*** Problematic Vertex shader source code ***

          #version 150
#define lowp
#define mediump
#define highp
#line 2
          in highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***
Switching to GLSL version 1.20, use the command line option '--useGLSL120' or '--useGLES'
```

Sometimes if breaks with segfault. I'll try to post the error when I get it again.

varna9000, 2023-03-20:
And here's an error, which showed when app broke:

```
Cannot make QOpenGLContext current in a different thread
[1]    22808 abort      ./OpenHantek
```

varna9000, 2023-03-20:
Thanks for replying.
Below is the output with `--useGLES`. In fact the program window opens blank and stays like this. It's not crucial and might be probably mac os QT related, but I just wanted to share my experience. and someone could probably give more clues to make it stable on arm64. I forgot to mention I'm using Mac OS Ventura.

Here's the terminal output:

```
./OpenHantek --useGLES
qt.qpa.fonts: Populating font family aliases took 78 ms. Replace uses of missing font family "Arial" with one that exists to avoid this cost.
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QPainter::begin: Paint device returned engine == 0, type: 1
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::font: Painter not active
QPainter::setFont: Painter not active
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
[... 33 lines cut ...]
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QPainter::begin: Paint device returned engine == 0, type: 1
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::font: Painter not active
QPainter::setFont: Painter not active
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
```

closed 2023-04-27 by github-actions[bot]

astelmashenko, 2025-01-17:
If I build with 
```
cd ~/Downloads/OpenHantek6022
git submodule update --init --recursive
rm -rf build/
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH=$(brew --prefix qt5)
make -j4

cd openhantek

$(brew --prefix qt5)/bin/macdeployqt OpenHantek.app -always-overwrite -verbose=2

python3 ../../utils/macdeployqtfix/macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek $(brew --prefix qt5)
```
It does not work. I works only if I build 
```
cd ~/Downloads/OpenHantek6022
git submodule update --init --recursive
rm -rf build/
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH=$(brew --prefix qt5)
make -j4
```

And then run executable.