OpenHantek/OpenHantek6022 #360: No graphics output - windows 10 3.3.2.2

squelch41, 2023-03-22:
Openhantek loads but, with the 'scope and in demo mode, I just get a black window.
Console output as screenshot here:
[image]

Openhantek window screenshot:
[image]

Using Windows 10, core i5-4300U laptop 8gb ram and integrated gfx

squelch41, 2023-03-22:
Oh yes, forgot to say, I tried that (and -e which I think is equivalent?)
get the following error:
[image]

and a white window:
[image]

DaPa, 2023-04-03:
Both openhantek_3.3.2.2-1_win_x64 and latest openhantek_3.3.2.2-9-g383ecd0-1_win_x64 work fine here (on Windows 10 Pro, 21H2, x64):
[image]
[image]

sqliuchang, 2023-04-03:
I have the same issue. And my PC is surface pro 3 which is also i5-4300u with 8G ram.

Ho-Ro, 2023-04-03:
> Version 3.3.2 is fine.

@gitguest0 3.3.2 used MSVC. Could this issue be a regression from the switch to mingw64?

gitguest0, 2023-04-03:
Probably, seems like a problem with very old Intel graphics drivers.
I just tested it on a Intel 7th gen Windows Tablet with no issues.

closed 2023-05-09 by github-actions[bot]

fabi9163, 2024-11-14:
I've the same bug, also Surface pro 3...is there now the required versio 5.15.11 available to build a new release which maybe would fix this?