OpenHantek/OpenHantek6022 #363: Program stops on Windows 11 when changing horizontal time base (In demo mode and connected with 6022BE)

louismulder, 2023-04-06:
When I want to change the timebase (Up or Down) the program stops.

Ho-Ro, 2023-04-07:
I don't understand people that ask for support, but are not willing to help the supporters.
When you open a new issue please select the correct button
[image]

It presents you with a template (see below) where you need to enter the **important information** (e.g. version, error mode, error messages etc.) - just leave the irrelevant questions open. Please mind especially the last section about **Additional context**, also on Windows you can use the command line.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
**No support for non-Linux related issues unless a volunteer steps in!**

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Screen resolution, screen setup (e.g. multimonitor, etc.)
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device [e.g. *Hantek 6022BE, ...*]
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

louismulder, 2023-04-07:
Hi Martin,

thx for your comment.
But maybe is the template too complex for a simple error. 
My OS is Windows 11.
The version of OpenHantek is: 3.2.2 in demo mode

Ho-Ro, 2023-04-07:
> The version of OpenHantek is: 3.2.2 in demo mode

[image]

This version is almost 2 years old - wontfix

louismulder, 2023-04-07:
Sorry Martin,

version is 3.3.2.2

Ho-Ro, 2023-04-07:
> But maybe is the template too complex for a simple error.

Just fill the necessary parts and leave the other open, I would be interested how the *"program stops"* - stop trace or freeze or crash? 
As mentioned, the additional context is important for hunting down an error, especially when it happens to the stable release that had a lot of downloads so far (more than 50% of the downloads are the Win version).
[image]

Please start the program from the cmd.exe window and report the error message. If this does not give a good hint, increase the verbosity to `OpenHantek --verbose 4`. This will give a lot of text, please copy/paste.

louismulder, 2023-04-07:
When I want to change change the timebase, then the following error occur:
QObject::setParent: Cannot set parent, new parent is in a different thread

Ho-Ro, 2023-04-09:
OMG, is it too much to ask for a little more info. This error does not occur in my tests, nor do many other Win users seem to have a problem with it. I would like to find out where your system behaves differently, unfortunately my crystal ball is under repair. Therefore, I choose the more cumbersome way for you to report as much as possible, including the output when the program was started with `OpenHantek --verbose 4`.
Otherwise you are free to return the program, ask for your money back and use the original Hantek software delivered together with your scope - SCNR.

koentjess, 2023-05-30:
Hey, maybe this helps. I am having the same problem in windows 11. Changing the compatibility settings to windows 7 or 8 fixes the crashing but the software does feel kinda slow when using it in compatibility mode. Using an older version of OpenHantek6022 works better for me. Idk if https://github.com/OpenHantek/OpenHantek6022/issues/111 has anything to do with it but the version that was suggested "3.1.2" works fine for me.

koentjess, 2023-05-31:
I see that the issue has been reopened. I have tested a few more version and the only versions that are giving problems are 3.3.2.2 and "unstable". I have added the output from the console as you mentioned above. One thing I found is that when running it in compatibility mode, the "oscilloscope lines" seem to be updating fine but when moving the cursor over some of the buttons on the right side, so the buttons under "voltage", "Horizontal", "Trigger" and "Spectrum"), the whole program seems to almost freeze for a moment.

Version: 3.3.2.2 - build: Mar  8 2023 17:36:56
12 ms: create openHantekApplication
22 ms: load translations for locale "en_NL"
25 ms: init libusb
30 ms: show splash screen
 FindDevices::FindDevices()
  FindDevices::updateDeviceList()
  FindDevices::updateDeviceList()
  FindDevices::updateDeviceList()
2940 ms: use device "DEMO" serial number "0000"
2944 ms: create DSO control thread
 HantekDsoControl::HantekDsoControl()
  Calibration data: "DEMO_0000_calibration.ini"
  HDC::getCalibrationFromEEPROM()
2958 ms: create settings object
 DsoSettings::DsoSettings() "DEMO" "0000" false
 DsoSettings::load() "\\HKEY_CURRENT_USER\\Software\\OpenHantek\\DEMO_0000"
2975 ms: create exporters
2977 ms: create post processing objects
 SpectrumGenerator::SpectrumGenerator()
 GraphGenerator::GraphGenerator()
2983 ms: setup OpenGL
3155 ms: set QFont(Arial,10,-1,5,50,0,0,0,0,0)
3158 ms: create main window
 MainWindow::MainWindow()
 VoltageDock::VoltageDock()
  VDock::loadSettings()
  VDock::setCoupling() 0 "DC"
  VDock::setGain() 0 "2 V"
  VDock::setUsed() 0 true
  VDock::setAttn() 0 1
  VDock::setInverted() 0 false
  VDock::setCoupling() 1 "DC"
  VDock::setGain() 1 "2 V"
  VDock::setUsed() 1 true
  VDock::setAttn() 1 1
  VDock::setInverted() 1 false
  VDock::setMode() "CH1 + CH2"
  VDock::setGain() 2 "2 V"
  VDock::setUsed() 2 false
  VDock::setAttn() 2 1
  VDock::setInverted() 2 false
 HorizontalDock::HorizontalDock()
  HDock::setSamplerate() 1e+06
  HDock::setTimebase() 0.001 return 0.001
  HDock::setFormat() 0
  HDock::setCalfreq() 1000
 TriggerDock::TriggerDock()
  TDock::loadSettings()
  TDock::setMode() 0
  TDock::setSlope() 0
  TDock::setSource() 0
  TDock::setSmooth() 0
 SpectrumDock::SpectrumDock()
  SDock::loadSettings()
  SDock::setMagnitude() 0 20
  SDock::setUsed() 0 false
  SDock::setMagnitude() 1 20
  SDock::setUsed() 1 false
  SDock::setMagnitude() 2 20
  SDock::setUsed() 2 false
  SDock::setFrequencybase() 1000
 GLScope::GLScope()
 GLScope::GLScope()
 DsoWidget::DsoWidget()
  DsoWidget::setupSliders()
  DsoWidget::setupSliders()
  VDock::loadSettings()
[... 274 lines cut ...]
   sampleIndex: 0 sRate: 200 sRate*duration: 4
   sampleIndex: 1 sRate: 500 sRate*duration: 10
   sampleIndex: 2 sRate: 1000 sRate*duration: 20
   sampleIndex: 3 sRate: 2000 sRate*duration: 40
   sampleIndex: 4 sRate: 5000 sRate*duration: 100
   sampleIndex: 5 sRate: 10000 sRate*duration: 200
   sampleIndex: 6 sRate: 20000 sRate*duration: 400
   sampleIndex: 7 sRate: 50000 sRate*duration: 1000
   sampleIndex: 8 sRate: 100000 sRate*duration: 2000
   sampleIndex: 9 sRate: 200000 sRate*duration: 4000
   sampleIndex: 10 sRate: 500000 sRate*duration: 10000
   sampleIndex: 11 sRate: 1e+06 sRate*duration: 20000
   sampleIndex: 11 sRate: 2e+06 sRate*duration: 40000
   sampleIndex: 11 sRate: 5e+06 sRate*duration: 100000
   sampleIndex: 11 sRate: 1e+07 sRate*duration: 200000
   sampleIndex: 11 sRate: 1.2e+07 sRate*duration: 240000
   sampleIndex: 11 sRate: 1.5e+07 sRate*duration: 300000
   sampleIndex: 11 sRate: 2.4e+07 sRate*duration: 480000
   sampleIndex: 11 sRate: 3e+07 sRate*duration: 600000
   sampleIndex: 11 sRate: 4.8e+07 sRate*duration: 960000
  HDC::controlSetSamplerate() 11 id: 10
  HDC::setRecordTime() emit samplerateChanged 500000
  HDock::setSamplerate() 500000
  DsoWidget::updateMarkerDetails()
  SDock::setSamplerate() 500000

koentjess, 2023-05-31:
**Describe the bug**
Program crashes when changing the "Timebase" in windows 11

**To Reproduce**
Start OpenHantek6022 version 3.3.2.2
Increase or decrease the Timebase by clicking any of the 2 buttons
Sometimes the program does not crash and lets you change it once, sometimes it instantly crashes
The crashing can be that the program does not respond for 10 seconds before quitting or just almost instantly quitting.

**Screenshots**
I dont think a screenshot really helps here

**Computer environment (please complete the following information):**
OpenHantek version: 3.3.2.2 or "unstable"
OS: Windows 11
Distribution, version: Windows 11 21H2
Screen resolution: 1920x1080 1 monitor Scale 125%, also crashes on 100%
Video hardware: Intel i7 10750H Igpu
OpenGL version: 4.6
Qt version: Don't think I need this when not compiling
C++ compiler version: Don't think I need this when not compiling

**Device [e.g. Hantek 6022BE, ...]**
Demo mode or Hantek 6022BL

**Additional context**
Versions 3.3.2.1 and below are fine

ChristopheMineau, 2023-06-10:
Hi, just updated to version 3.3.2.2.1 and I wanted to mention that I have the exact same problem as described by koentjess , except that I personally am on Windows 10.
The last traces on the terminal are exactly the same.
I noticed that if I start it without verbose, it lets me change the Timebase once, but if I add --verbode 4 , it crashes at the first attempt to change the timebase.
Maybe it's a coincidence.
So nothing particular displayed at all on the screen, it freezes maybe for half a second and then the window disappears.

In the Windows journal, I can see:
Name of the failing application:  OpenHantek.exe, version : 0.0.0.0, horodatage : 0x6408c7e0
Name of the failing module : OpenHantek.exe, version : 0.0.0.0, horodatage : 0x6408c7e0
Exception code : 0xc0000005
Error offset : 0x0000000000e99827
Failing process Id : 0x47ec
Starting time : 0x01d99b6b9c021415
Failing app path : C:\INSTALLATIONS\OpenHantek\OpenHantek6022_v3.3.2.2\OpenHantek.exe
Failing module path: C:\INSTALLATIONS\OpenHantek\OpenHantek6022_v3.3.2.2\OpenHantek.exe
Report Id : 9ba89d1f-2ef4-437a-8220-8ab18ef5efb4
Complete name of the failing package : 
Id of the application relative to the failing package :

Ho-Ro, 2023-06-19:
> PS : previous version 3.3.2.1 works for me, it must be a degradation in 3.3.2.2

Looking at the [changes between 3.3.2.1 and 3.3.2.2](https://github.com/OpenHantek/OpenHantek6022/compare/3.3.2.1...3.3.2.2) I suspect that [1d11253](https://github.com/OpenHantek/OpenHantek6022/commit/1d11253ce829346e2758e97cabcccd83824543dc) has the biggest impact. I need support from a windows developer who could build both variants (msvc and mingw64) and check - I do not use windows and cannot investigate.

nseam, 2023-08-07:
@Ho-Ro I've just built and run the project on Windows 11 (took me ~2h to configure all the things) and can confirm that just by increasing or decreasing the Timebase, there is a problem with non-main-thread accessing widgets created by main thread. I think the widget callbacks should delegate somehow its execution to the main thread. I am a little newbie in Qt and will receive my first oscilloscope tomorrow :D It will be Hantek 6022BL, so I thought I can work on the self-built version and help on any further issues.

[image]

If this is not something you can easily fix then I can work on this. Just need to go deeper how Qt, callbacks and threading works.

commit 875a8dc9 (2023-08-08): Fixes #363, refs #365. Splitted HorizontalDock::timebaseChanged event handler from doing operations on dsoControl and dsoWidget in the same callback. Now there two callback and each uses the Widget it operates on.
closed 2023-08-08 by Ho-Ro