OpenHantek/OpenHantek6022 #364: MacOS Monterey Firmware Upload Error

alexjoyner, 2023-04-07:
Trying to OpenHantek for a 6022BE model scope and got this error

Connection Failed!
UploadFirmware: libusb_claim_interface() failed: LIB_USB_ERROR_NOT_FOUND
[image]

OS: MacOS Monterey V12.0.1

Ho-Ro, 2023-04-07:
> Once the program loaded successfully, but only after several attempts

Then the USB access rights for the Hantek should be ok. Some Ideas that come to my mind:

- Sometimes other devices on USB can disturb the communication, try unplugging all other devices, especially webcam, TV/SDR stick or Logic analyser.
- You could try another USB cable or another USB port. It was reported that USB3 could be unstable, so better use an USB2 port - if not available you could also try an cheap external USB2 hub.
- If this doesn't help try to start the program from a termial window with `OpenHantek --verbose 4` and report the output.

alexjoyner, 2023-04-07:
Thanks @Ho-Ro!

So I was able to get the program running, but I'm not sure I would call this closed yet.

The app opened more consistently if I closed OpenHantek, then plugged in the USB and re-launched the program after everything was connected. This is opposed to what I was doing before which was running the program, then connecting the device and selecting it from the dropdown. However; I would still sometimes get the error listed before, and sometimes I would get this new error:

[image]

Here's a short video of the finickiness I would run into from closing and re-opening the program with the USB connected:
[image]

I tried this using several different cables and got the same results.

Also, a quick note if others run into this:
> If this doesn't help try to start the program from a termial window with OpenHantek --verbose 4 and report the output.
For this to work, I think I would have to set up a path to OpenHantek because I get OpenHantek command not found. For me it was easier to run: `open -n ./OpenHantek.app --args  --verbose 4`
Note: I haven't run the open command from the terminal in a while and this might not have picked up the verbose arg because I got no output.

Ho-Ro, 2023-04-08:
I propose to connect the device and then start the program - this is at least my typical use case. The FW upload is only needed once, successive program starts after closing the program just start with FW already in place.

closed 2023-05-15 by github-actions[bot]

stalegjelsten, 2023-05-20:
Hi! I just wanted to say that I also have the same problem. OpenHantek6022 worked flawlessly for me in February of this year. At that point I was on macOS 12.6 Monterrey. 

I have since upgraded to macOS 13.3 Ventura, and now I have trouble connecting my Hantek6022BL via USB. Previously, the OpenHantek software recognized my oscilloscope immediately, even though it was plugged into my –admittedly overcrowded– Thunderbolt3 dock. Now I was only able to get a connection after rebooting my computer without any peripherals plugged in, and connecting directly to USB port (I still had to use a simple tiny USB-A to USB-C adapter since the MacbookPro doesn't come with any USB-A ports). 

The red LED doesn't come on when I try to connect the oscilloscope except on one occasion when it was blinking. It still didn't connect though. The LED  works when I put it in LA-mode and connect to PulseView.