OpenHantek/OpenHantek6022 #385: [Q] Calibration

Sineos, 2023-10-25:
Since quite some time I performed a calibration (50 Ohm terminators) according to the manual and the`DSO-6022BL_C8B39B296C51_calibration.ini` is indeed newly created but contains only gain values of 1 offsets of 0. 

Is this possible?

Env:
Windows: 10.0.19045.3570
Device: 6022BL
Build: 3.3.3-12-gd5ef54e-unstable

EEPROM calibration data:
```
C:\Users\souls\Documents\GitHub\Hantek6022API>.\examples\get_calibration.py
0x86 0x87 0x86 0x87 0x86 0x87 0x83 0x85 0x81 0x84 0x7f 0x83 0x7f 0x83 0x7f 0x83 0x86 0x87 0x86 0x87 0x86 0x87 0x83 0x85 0x81 0x83 0x7f 0x82 0x7f 0x82 0x7f 0x82 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 0x80 
```

Sineos, 2023-10-26:
Additional information:
Running
```
C:\Users\souls\Documents\GitHub\Hantek6022API>.\examples\calibrate_6022.py -c
Setting up scope...

Calculate zero adjustment
Apply 0 V to both channels and press <ENTER>
Measure offset at low speed for gain  10
Measure offset at high speed for gain  10
Measure offset at low speed for gain  5
Measure offset at high speed for gain  5
Measure offset at low speed for gain  2
Measure offset at high speed for gain  2
Measure offset at low speed for gain  1
Measure offset at high speed for gain  1
Traceback (most recent call last):
  File "C:\Users\souls\Documents\GitHub\Hantek6022API\examples\calibrate_6022.py", line 170, in <module>
    product = scope.get_product_string()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\PyHT6022\LibUsbScope.py", line 271, in get_product_string
    return self.device.getProduct()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 2003, in getProduct
    return self.open().getProduct()
           ^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 2055, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]
^C
C:\Users\souls\Documents\GitHub\Hantek6022API>
```

while this one works:
```
C:\Users\souls\Documents\GitHub\Hantek6022API>.\examples\calibrate_6022.py -c
Setting up scope...

Calculate zero adjustment
Apply 0 V to both channels and press <ENTER>
Measure offset at low speed for gain  10
Measure offset at high speed for gain  10
Measure offset at low speed for gain  5
Measure offset at high speed for gain  5
Measure offset at low speed for gain  2
Measure offset at high speed for gain  2
Measure offset at low speed for gain  1
Measure offset at high speed for gain  1
Traceback (most recent call last):
  File "C:\Users\souls\Documents\GitHub\Hantek6022API\examples\calibrate_6022.py", line 170, in <module>
    product = scope.get_product_string()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\PyHT6022\LibUsbScope.py", line 271, in get_product_string
    return self.device.getProduct()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 2003, in getProduct
    return self.open().getProduct()
           ^^^^^^^^^^^
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 2055, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "C:\Users\souls\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb1\__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]
^C
C:\Users\souls\Documents\GitHub\Hantek6022API>
```

I tried different `libusb-1.0.dll` versions but this did not change anything.

Ho-Ro, 2023-11-05:
You could change in `calibrate_6022.py` the lines from
```python
    product = scope.get_product_string()
    if product:
        calFile = product + "_"
    else:
        calFile = "DSO-6022BE_"
    serial_number = scope.get_serial_number_string()
    if serial_number:
        calFile += serial_number + "_calibration.ini"
    else:
        calFile += "_NN_calibration.ini"
```
to 
```python
    calFile = "DSO-6022BL_NN_calibration.ini"
```
and retry calibrating, dunno why Windows doesn't get the correct product string (and maybe also the serial number).

Sineos, 2023-11-05:
Many thanks @Ho-Ro, this works.

The resulting file does look a bit strange. Maybe you have an idea:
```
;OpenHantek calibration file: DSO-6022BL_NN_calibration.ini
;Created by tool 'calibrate_6022.py'

[offset]
ch0\20mV=  5.46
ch0\50mV=  5.46
ch0\100mV=  5.46
ch0\200mV=  3.00
ch0\500mV=  0.89
ch0\1000mV= -1.00
ch0\2000mV= -1.00
ch0\5000mV= -1.00
ch1\20mV=  6.52
ch1\50mV=  6.52
ch1\100mV=  6.52
ch1\200mV=  5.02
ch1\500mV=  3.41
ch1\1000mV=  2.53
ch1\2000mV=  2.53
ch1\5000mV=  2.53

[gain]
ch0\20mV=1.0
ch0\50mV=1.0
ch0\100mV=1.0
ch0\200mV=1.0
ch0\500mV=1.0
ch0\1000mV=1.0
ch0\2000mV=1.0
ch0\5000mV=1.0
ch1\20mV=1.0
ch1\50mV=1.0
ch1\100mV=1.0
ch1\200mV=1.0
ch1\500mV=1.0
ch1\1000mV=1.0
ch1\2000mV=1.0
ch1\5000mV=1.0

[eeprom]
replace_eeprom=true
```

Ho-Ro, 2023-11-05:
These lines? Why not, the offset can be positive or negative. The Hantek has only 4 different gain steps, x10 for 20, 50, 100 mV; x5 for 200 mV; x2 for 500 mV and x1 for 1, 2, 5 V - so the 1st three offset values are identical, same for the last three ones.

```
ch0\1000mV= -1.00
ch0\2000mV= -1.00
ch0\5000mV= -1.00
```

Sineos, 2023-11-05:
Thanks for the explanation and support. 👍 
I was a bit surprised to see 3 times -1.00 and -2.53 in a row.

closed 2023-11-05 by Sineos