OpenHantek/OpenHantek6022 #388: No traces seen under v3.3.3

softfoot, 2023-12-13:
In demo mode (and normal running) of v3.3.3 I see no traces - all the text messages (measurements etc) are present - just NO traces:
v3.3.2.1 works correctly.
Altering trigger points and other settings doesnt help.
Running under Win7 PRO 64bit
Single monitor 1920/1080
Built in display on Laptop ThinkPad T410 (I think)
Reproduceable in DEMO mode

Ho-Ro, 2023-12-13:
@gitguest0 could you please check if your build system change can cause this issue.

Ho-Ro, 2023-12-17:
@softfoot you could check the msvc build from latest unstable release

Ho-Ro, 2024-01-24:
As @softfoot did not provide the requested info I suppose he's happy with the MSVC build. I will close now.

closed 2024-01-24 by Ho-Ro