OpenHantek/OpenHantek6022 #390: 'Version' field value 'devdrop-3.3.3-22-g1c3a681-dirty': version number does not start with digit

snyggapa, 2023-12-27:
possibly my lack of knowledge but trying to build this in a Linux (debian 11) host inside a chromebook

LinuxSetup_AsRoot worked fine
LinuxBuild worked fine

installing the .deb returns this below.

Is this a problem with the dev build and if so can the command line be forced to build the last "numeric" version 3.3.3 somehow - or am I missing something.

Any help appreciated 

myname@myhost:~/OpenHantek6022/build/packages$ sudo apt install ./openhantek_devdrop-3.3.3-22-g1c3a681-dirty_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'openhantek' instead of './openhantek_devdrop-3.3.3-22-g1c3a681-dirty_amd64.deb'
The following packages were automatically installed and are no longer required:
  bsdmainutils cpp-8 libapt-inst2.0 libcodec2-0.8.1 libcroco3 libcrystalhd3 libdns-export1104 libdouble-conversion1 libicu63 libidn11
  libigdgmm5 libip4tc0 libip6tc0 libiptc0 libisc-export1100 libisl19 libjson-c3 libllvm7 libmpdec2 libnftables0 libperl5.28
  libprocps7 libpython3.7-minimal libpython3.7-stdlib libreadline7 libvpx5 libx264-155 libx265-165 libxcb-util0 ncal
  perl-modules-5.28 python3.7-minimal usb.ids
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  openhantek
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/6,359 kB of archives.
After this operation, 371 kB disk space will be freed.
Get:1 /home/steveb/OpenHantek6022/build/packages/openhantek_devdrop-3.3.3-22-g1c3a681-dirty_amd64.deb openhantek amd64 devdrop-3.3.3-22-g1c3a681-dirty [6,359 kB]
debconf: delaying package configuration, since apt-utils is not installed
dpkg: error processing archive /home/steveb/OpenHantek6022/build/packages/openhantek_devdrop-3.3.3-22-g1c3a681-dirty_amd64.deb (--unpack):
 parsing file '/var/lib/dpkg/tmp.ci/control' near line 10 package 'openhantek':
 'Version' field value 'devdrop-3.3.3-22-g1c3a681-dirty': version number does not start with digit
Errors were encountered while processing:
 /home/steveb/OpenHantek6022/build/packages/openhantek_devdrop-3.3.3-22-g1c3a681-dirty_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

snyggapa, 2023-12-28:
I managed a workaround by changing line 29 on CMakelists.txt to

       set( VERSION "${VERSION}-devdrop" )

not sure if this is the "correct" fix or not but it worked. Now I am just fighting getting the USB device to be seen on my chromebook..

closed 2023-12-28 by Ho-Ro

Ho-Ro, 2023-12-28:
Thx, fixed it

commit b5e1aee0 (2023-12-28): fix #390