OpenHantek/OpenHantek6022 #391: I want to print 1 or 0 on the bottom of each wave form

quantrpeter, 2024-01-07:
Hi
   I want to print 1 or 0 on the bottom of each wave form. Please tell me where i should modify? Should i just add my code to void GlScope::drawVoltageChannelGraph ?  How to get the value of each point? I can't dump out all values from these:

Graph::VaoCount &v = graph.vaoVoltage[ channel ];

thanks
Peter

quantrpeter, 2024-01-07:
ok, i can dump out all points in wave form here. Please tell me how to do next, thanks

[image]

Ho-Ro, 2024-01-08:
Please explain what you want to achieve. If you want the measured values as a list you can export them as csv or json from the export menu. If you want to hack your own format take this as a starter.