OpenHantek/OpenHantek6022 #404: Horizontal greyed out

chumariesco, 2024-05-06:
I'm not sure if this is a bug, but suddenly the section for Horizontal settings disappeared.  Why could this be?
Voltage, Trigger and Spectrum are there, but Horizontal is has an empty checkbox when I right click on that area and it cannot be activated.
It worked fine before and I didn't update it ASAIK. Using a Hantek 6022BE.

Ho-Ro, 2024-05-07:
Please provide a minimum of useful info that helps me to help you, e.g. program version, operating system, maybe a screen shot that you easily can create with menu `Export/Screenshot`:

[image]

An easy fix could be to start with a fresh setup - use menu `Oscilloscope/Settings/Scope`:

[image]

and tick `Apply default settings after next restart`:

[image]

Ho-Ro, 2024-05-13:
> Voltage, Trigger and Spectrum are there, but Horizontal has an empty checkbox when I right click on that area and it cannot be activated.

You can move or stack the four sections (the latter is particularly useful for very small screens). The horizontal section may have been inadvertently closed by this action. By ignoring the saved status and starting with a new configuration as described above, this problem can be easily solved.

I have not encountered this problem when moving and rearranging these sections intensively during my tests, so it would be helpful to know your exact configuration, i.e. operating system, graphics HW, program version. Since you don't provide any of these details, I can only assume that you are using Windows (like the majority of users). Since I don't use Windows productively and only run the program sporadically on a Windows computer for testing purposes, I can't do much without this exact information.

However, since there has been no feedback so far, I assume that the problem has now been solved and I am closing this issue. Please open again if required.

closed 2024-05-13 by Ho-Ro

chumariesco, 2024-07-12:
Thanks a lot Ho-Ro and sorry for the delay to answer. I'm just coming back to the issue and reverting to default settings as you suggested solved it. No idea how it happened, but it's easy to fix ;)