OpenHantek/OpenHantek6022 #406: macOS application does not use primary language

jamieshaw, 2024-06-24:
**Describe the bug**
I currently have two languages configured on my Mac — English (UK), and French — the former being the primary and system language.  When opening OpenHantek, the language is set to French.  Removing French from the language list causes the application to use English.

Steps to reproduce the behavior:
1. Go to macOS Settings > General > Region
2. Add a second language
3. Open OpenHantek

**Expected behavior**
Application should track primary system language rather than the (assumingly) the bottom-most in the System Language list.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: Version 3.3.3
 - OS: macOS
 - Distribution, version: Sonoma 14.5 (23F79)

**Scope device (please complete the following information):**
 - Device: Hantek 6022BE
 - Program top line: OpenHantek6022 (3.3.3) – DSO-6022BE (FW0210)
 - Input signal: N/A
 - Probe setting: N/A

Ho-Ro, 2025-03-03:
@jamieshaw Please check the latest pre-release [3.4-rc4](https://github.com/OpenHantek/OpenHantek6022/releases/tag/3.4-rc4) to see if the problem persists.

jamieshaw, 2025-03-05:
Off the bat, the app doesn't want to open and Apple's "Problem Reporter" pops up.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenHantek [52046]
Path:                  /Applications/OpenHantek.app/Contents/MacOS/OpenHantek
Identifier:            OpenHantek
Version:                (3.4-rc4)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-05 09:03:28.4077 +0000
OS Version:            macOS 15.3 (24D60)
Report Version:        12
Anonymous UUID:        5E1F86A5-7709-ECA8-6162-8385DC36A2B3

Sleep/Wake UUID:       205764AF-E431-4619-A84A-D3F47E08AFF8

Time Awake Since Boot: 770000 seconds
Time Since Wake:       332479 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x00000001005e8000
Exception Codes:       0x0000000000000032, 0x00000001005e8000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

VM Region Info: 0x1005e8000 is in 0x1005e8000-0x100640000;  bytes after start: 0  bytes before end: 360447
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT                  1005b8000-1005e8000    [  192K] r--/r-- SM=COW  /Applications/OpenHantek.app/Contents/MacOS/OpenHantek
--->  mapped file                 1005e8000-100640000    [  352K] r--/rwx SM=COW  Object_id=a57970a9
      GAP OF 0x27c000 BYTES
      mapped file                 1008bc000-100908000    [  304K] ---/--- SM=COW  Object_id=5d80fa35

Thread 0 Crashed:
0   dyld                          	       0x19ab0a220 dyld3::MachOFile::compatibleSlice(Diagnostics&, unsigned long long&, void const*, unsigned long, char const*, dyld3::Platform, bool, dyld3::GradedArchs const&, bool) + 84
1   dyld                          	       0x19aadfd18 invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 116
2   dyld                          	       0x19aae43b8 dyld4::SyscallDelegate::withReadOnlyMappedFile(Diagnostics&, char const*, bool, void (void const*, unsigned long, bool, dyld4::FileID const&, char const*) block_pointer) const + 132
3   dyld                          	       0x19aadfc70 dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 208
4   dyld                          	       0x19aad39c4 dyld4::Loader::makeDiskLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 172
5   dyld                          	       0x19aad522c invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 2556
6   dyld                          	       0x19aad4084 dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 780
7   dyld                          	       0x19aac365c dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, dyld3::Platform, bool, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 556
8   dyld                          	       0x19aad3bb8 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 264
9   dyld                          	       0x19aad42b0 dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 492
10  dyld                          	       0x19aaddbb0 invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 516
11  dyld                          	       0x19aaf9b28 invocation function for block in mach_o::Header::forEachLinkedDylib(void (char const*, mach_o::LinkedDylibAttributes, mach_o::Version32, mach_o::Version32, bool&) block_pointer) const + 136
12  dyld                          	       0x19aaf9364 mach_o::Header::forEachLoadCommand(void (load_command const*, bool&) block_pointer) const + 284
13  dyld                          	       0x19aaf9828 mach_o::Header::forEachLinkedDylib(void (char const*, mach_o::LinkedDylibAttributes, mach_o::Version32, mach_o::Version32, bool&) block_pointer) const + 160
14  dyld                          	       0x19aadd88c dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 164
15  dyld                          	       0x19aac10d4 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 1232
16  dyld                          	       0x19aac0bc0 dyld4::start(dyld4::KernelArgs*, void*, void*)::$_0::operator()() const + 544
17  dyld                          	       0x19aac005c start + 2304

Thread 0 crashed with ARM Thread State (64-bit):
[... cut ...]
USB Device: 4-Port USB 3.0 Hub
USB Device: USB-C Pro Card Reader
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: 4-Port USB 2.0 Hub
USB Device: C922 Pro Stream Webcam
USB Device: USB Receiver
USB Device: BillBoard Device
USB Device: LG Monitor Controls
USB Device: iPad
USB Device: CalDigit USB-C Pro Audio
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: USB-C Pro Dock, CalDigit, Inc., 1, 61.1
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

Ho-Ro, 2025-03-05:
How's the intel version?

jamieshaw, 2025-03-07:
> How's the intel version?

It runs, but it's still in French… 😅 
[image]

Ho-Ro, 2025-03-08:
Please check my "kind-of-work-around" solution from last commit:

    Menu->Oscilloscope->Scope->Use the international interface

Ho-Ro, 2025-03-08:
I do not use a Mac and cannot test your setup, but I have also two languages configured on my Linux, 1st German, 2nd (British) English and OpenHantek starts up in German unless started as `OpenHantek -i` (or `OpenHantek --international`), this option bypasses all localisation and uses the original English text strings. Or use the option described above.

Ho-Ro, 2025-03-08:
@jamieshaw do you have the toolchain to build OpenHantek from source code?
If yes, could you change [`main.cpp, line 143-144`]( https://github.com/OpenHantek/OpenHantek6022/blob/17d05f864666f26b76860865d08e954f2e0e3df0/openhantek/src/main.cpp#L143-L144) from

        if ( !doNotTranslate && useLocale &&
             QLocale().name() != "en_US" ) { // somehow Qt on MacOS uses the german translation for en_US?!

to

        if ( !doNotTranslate && useLocale &&
             QLocale().name().left(3) != "en_" ) { // somehow Qt on MacOS uses the german translation for en_US?!

and see if it works?

This hack was suggested by a Mac user many years ago. I suspect that it should be extended to all English-speaking countries.

closed 2025-03-09 by Ho-Ro

Ho-Ro, 2025-03-10:
I implemented the proposed change -> no translation if the locale is "en_*", this covers all english-speaking countries now.

commit e745ffa4 (2025-03-09): try to fix #406, no UI translation now for LANG=en_* (not only en_US)
commit f82859fc (2025-03-09): try to fix #406, no UI translation now for LANG=en_* (not only en_US)