OpenHantek/OpenHantek6022 #411: rules file

artble, 2024-11-01:
**Describe your intention**
A clear and concise description of your idea

when i ran the program i got
Gtk-Message: 20:21:18.318: Failed to load module "xapp-gtk3-module"
Qt: Session management error: Could not open network socket

also the app showed me (translatet from my native language)
"make sure that you copied file of rules udev to /etc/udev/rules.d/60-openhantek.rules or /usr/lib/udev/rules.d/60-openhantek.rules to get proper access premitions to USD. 
All i have is demo mode, how do i get normal mode which actually works?

artble, 2024-11-01:
also after i chose my device DSO-6022BE, there was a messege "connection failed! UploadFirmware: Couldn't open device: Access denied (insufficient premissions)

Ho-Ro, 2024-11-02:
# Bug report - please fill the relevant info to help me to help you

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
**No support for non-Linux related issues unless a volunteer steps in!**

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Screen resolution, screen setup (e.g. multimonitor, etc.)
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device [e.g. *Hantek 6022BE, ...*]
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

closed 2025-02-19 by Ho-Ro