OpenHantek/OpenHantek6022 #417: UI freeze when right-clicking

dzid26, 2025-02-05:
**Describe the bug**
When time base is 200us or lower and I right click on the plot, the tooltip glitches, the framerate drops and within a few seconds the app becomes unresponsive and never recovers.  

It's anno

**To Reproduce**

Steps to reproduce the behavior:
1. Use 1khz calibration out and connect the probe to it 
2. Set the trigger on the squarewave, e.g rising edge
3. Set the timebase to 10us 
4. Right click on the plot

**Expected behavior**
Tooltip should show up

**Screenshots**
Before the freeze:
[image]

**Computer environment (please complete the following information):**
 - OpenHantek version: [*release v3.3.3  and 3.4-rc2-15-gffd1691*]
 - OS: [*windows11*]
 - Screen resolution, screen setup (fullhd 60Hz and 4k 60Hz)
 - Video hardware: [ *Intel Iris Xe*]

**Scope device (please complete the following information):**
 - Input signal: [ *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Ho-Ro, 2025-02-05:
3.3.3 is old and has issues with some graphic constellations on Windows, please try one of the two Windows packages from the [latest release 3.4-rc3](https://github.com/OpenHantek/OpenHantek6022/releases/tag/3.4-rc3):
https://github.com/OpenHantek/OpenHantek6022/releases/download/3.4-rc3/openhantek_3.4-rc2-15-gffd1691_mingw_x64.zip
https://github.com/OpenHantek/OpenHantek6022/releases/download/3.4-rc3/openhantek_3.4-rc2-15-gffd1691_msvc_x64.zip
If this does not help I cannot do anything because I do not use Windows and cannot debug your issues.

dzid26, 2025-02-05:
None of those helped. 

Just to confirm, you don't experience that on Linux?

https://github.com/user-attachments/assets/a6ed157a-08c9-44bc-b520-1ee759d52e1a

When the FPS is high, it seems that the tooltip doesn't have time to fully draw and eventually some race condition occurs. It eats up one and then four cores. Ram also slowly increases and fluctuates 100-500MB. 
After 3min it does eventually unblock itself.

Setting _**Minimal Time between capture frames**_ to 20ms is a workaround. Presumably because 20ms > 1/60Hz.

Ho-Ro, 2025-02-06:
> Just to confirm, you don't experience that on Linux?

No, never saw this kind of freeze. I tried very hard to reproduce just some minutes ago - nothing, sorry.