OpenHantek/OpenHantek6022 #418: Screen right side voltage, horizontal, trigger docks missing

KLR685, 2025-02-19:
Openhantek6022 (3.3.3) - device DSO-6022BE (FW0210)

HP Elitebook, Windows 10 pro x64

OpenHantek HAS been running successfully for some time so probably user (me) induced error but I cannot recover from it

Other than the "spectrum" dock, the others that should be at the right hand side of the screen (voltage, horizontal, trigger) are missing. Right clicking on the top bar of the Spectrum dock shows toolbar and specturm ticked, the others not. Toolbar can be selected/deselected and and the left side toolbar shows/hides as expected. The others, including spectrum, are greyed out and cannot be selected.
I have searched the hard drive for any instance of "openhantek" and removed it. Reinstalling returned to exactly the same fault/condition as described above..
I have installed OpenHantek on a desktop computer and with the 6022 connected it shows the four docks as expected. Connecting the laptop to the desktop lan, connecting the 6022 to the laptop, opening the desktop Openhantek folder from the laptop and running openhantek.exe on the laptop gives me the missing docks. Warm boot, cold boot, replugging the 6022 etc make no difference. Where else on the laptop hard drive could anything relating to open hantek be hiding?

Ho-Ro, 2025-02-19:
Please check if the solution for #404 works for you.

KLR685, 2025-02-20:
Thank you for the prompt response. I tried the "apply default settings after next restart" both from the gui and also from the command line, neither had any effect  ie no change.
The Spectrum box can be dragged to the left out into the middle of the screen, or dragged back right, to show a full height area (where all the docks would be) but I cannot find any way of getting the missing ones back. Are there any Windows registry settings associated with OpenHantek? 
thanks again
Peter

Ho-Ro, 2025-02-20:
>  Are there any Windows registry settings associated with OpenHantek?
Here's documented where Qt stores its persistent data:
https://doc.qt.io/qt-5/qsettings.html#locations-where-application-settings-are-stored
Unfortunately I cannot support b/c I do not use any Windows and I have no access to my Hantek at the moment. So you're on your own unless a kind Win user can support :(

KLR685, 2025-02-23:
Good afternoon Martin, I did have a poke around the registry looking for QSettings to no avail, but looking into Hkey_Current_User\Software showed an entry for OpenHantek with various subkeys. After backing up the registry I deleted the entire OpenHantek key (What could possibly go wrong?), connected the Hantek box and...it works! I have my docks back. So all is good. Thank you for taking the time

closed 2025-02-23 by KLR685