OpenHantek/OpenHantek6022 #422: Display internal details (oversampling, hardware gain)

Sogolumbo, 2025-03-03:
**Is your feature request related to a problem?.**
The SNR of the oscilloscope varies based on the settings. This is confusing at the beginning. I'm pretty sure this is due to the oversampling and hardware/software gain.

**Describe the solution you'd like**
Display the oversampling factor and the hardware/software gain. This will stop users from wondering about the varying SNR and will make it easier to achieve high SNR when necessary.

Sogolumbo, 2025-03-03:
I have one more idea which relates to this:
It would be nice to calculate the quantization error in absolute units.
Based on that it is possible to calculate the signal to quantization error ratio (SQNR) and thus the minimum THD which can be measured at the current scope settings and the signal level provided.

One way to use this result would be to display "≤" in front of the THD reading when it reaches the SQNR.

References:
[SQNR calculation](https://dsplog.com/2007/03/19/signal-to-quantization-noise-in-quantized-sinusoidal/)

Ho-Ro, 2025-03-03:
Interesting project, I will keep this in mind for the next version 3.5. Due to lack of time I cannot implement it at the moment, but I would ask you to provide a proof of concept or better a PR.

Ho-Ro, 2025-03-04:
Oversampling display was added with 7b8942a.

> what the chances are for the trigger to miss or hit based on the current settings

The triggering is the weakest point of the Hantek HW (it's non-existing). I do it in SW, as described [there](https://github.com/OpenHantek/OpenHantek6022/issues/214#issuecomment-906489194):

> Due to the missing trigger HW of the scope OpenHantek6022 is best suited for **repetitive** signals in the **extended audio range** (1 Hz .. 100 kHz). If you want to catch single events reliable - get another hardware.
A DSO works like this: capture the input signal into a buffer (e.g. for 20 ms/div * 10 div = 200 ms) and when the buffer is filled refresh the screen with the new buffer content. This would give one refresh every 200 ms for a free-running signal. But to get a standing display you want to trigger. The Hantek6022 don't have any hardware trigger, so triggering is done completely in SW. Therefore the program samples (free running) into a bigger buffer that equates twice the screen size, searches for a sufficient trigger point and displays the corresponding section of the buffer. This leads to a refresh every 400 ms. If there is no trigger condition the sampling starts again until you catch the slope. You should check [how it is done](https://github.com/OpenHantek/OpenHantek6022/blob/165f31ef308feaa833475c4e820a827286b148ca/openhantek/src/hantekdso/hantekdsocontrol.cpp#L571).

closed 2025-03-04 by Ho-Ro

Sogolumbo, 2025-03-04:
I read that part but I still don't understand everything:

> Therefore the program samples (free running) into a bigger buffer  

What does free running actually mean? Does the oscilloscope just send all the data?

The entire point of my questions is to find out:
1. which percentage (in the time domain) of the whole signal can be displayed for a given setting (in auto mode)
2. does the probability of the trigger not missing the signal equal to the percentage above?

My naive assumption for short time windows is that only part of the signal is shown because at some point part of the signal is discarded.
Is the 1ms delay rule in the settings the only reason that signals get discarded or are there more?

I'll introduce some math:
`w`: time span shown (width of the window, 200ms in the example above)
`W = 2*w`: time span captured (width of the data, 400ms in the example above)
`fW = 1/W`: refresh rate of the buffer if no sample where to be dropped
`f = 1/T`: refresh rate of the buffer
`T`: how long each buffer is the current one

So 
`W/T` is the percentage of the signal which is shown and 
`1-W/T` is the percentage of the signal which is discarded.

Now what is T?
Can it be calculated or does it depend on system performance? If it depends on system performance it can just be measured.
It will depend on the 1ms rule, right?

Is T the same in `auto`, `normal` and `once` mode?
It should be if it doesn't depend on system performance and normal and once only change if the current buffer is shown on screen or not.

Sogolumbo, 2025-03-04:
> Enough values left and right of the trigger to get a full screen for the given trigger position.

This cuts the probability of a successful trigger in half.
So in a best case scenario we have:
`h = 1ms`: holding time (settings)
`p = 0`: processing time (unknown)
`T = W+h+p`: duration of each buffer
`P = W/T/2`: probability of a successful trigger on a singular event.

If the processing time `p` is neglected we end up with this plot:

[image]

The maximum probability is strictly determined by `Pmax = 1-w/W` (The bigger the buffer compared to the displayed content, the higher the probability).
The position of the dropoff is at `p+h≈W = W/w*10*b`.

So both values can in theory be improved by having a bigger buffer size compared to the width of the window.

Sogolumbo, 2025-03-04:
This is the python code for the plot:

```python
import numpy as np
from matplotlib import pyplot as plt
# all numbers in SI units (e.g. seconds) unless specified otherwise

scale = np.array([1, 2, 5])
b = scale # time basis (5 s down to 1 ns)

for x in [1e-1, 1e-2, 1e-3, 1e-4, 1e-5, 1e-6, 1e-7, 1e-8, 1e-9]:
    b = np.append(scale*x, b)

W = 20*b # buffer size in time
h = 1e-3 # holding time (settings)
p = 0 # processing time (unknown)
T = W+h+p # duration of each buffer
P = W/(T)/2 # probability of a successful trigger on a singular event

#plt.plot(b, W)
#plt.plot(b, T)

plt.plot(b, P*100)
plt.grid()
plt.xscale('log')
plt.ylabel('probability in %')
plt.xlabel('time base in s')
plt.suptitle('Probability of a successful trigger on a singular event')
plt.title('(Neglecting processing time)')
plt.show()
```

Sogolumbo, 2025-03-04:
How is the ratio of `w/W` defined in code? I think I would like to experiment with this in the future to get a better responsiveness. I found `SAMPLESIZE` in `dsosamples.h`. But how is the amount of samples on the screen defined?

It seems likely to me that a maximum probability between 75% and 95% (`W/w` betwen `4` and `20`) is achievable. This would be a noticeable improvement if you ask me.

Sogolumbo, 2025-03-04:
Another way to achieve an improvement would be to loosen the condition:
> - Enough values left and right of the trigger to get a full screen for the given trigger position.
That would not be very elegant though.

Sogolumbo, 2025-03-04:
Ooh... I have yet another idea which seems pretty elegant to me. But it has an obvious problem which I missed when it first came into my mind:
It all boils down to continuous data capture. If there's no gap between following blocks that would make it possible to put two following blocks of the window size (10kS) as one block. The blocks get put into a FiFo queue of length 2. This would achieve 100% trigger probability while halving the response time compared to the current approach.

It seems like it's somewhat similar to the rolling mode.

Sogolumbo, 2025-03-05:
`w/W=10` seems to be a good compromise between high success rate and low additional delay.
For very slow signals the roll mode is more useful because it doesn't introduce noticeable delay.
If anybody needs 100% success rate they have to record the signal, there's no way around that.

[image]