OpenHantek/OpenHantek6022 #427: Calibration not saved; gain values all 1, offset all 0

redfast00, 2025-03-28:
When I try to calibrate the scope according to the procedures in the README, it is not persistent as I expected: I have to do this every time. When I look at the calibration file, I see that the gain values are all 1, and the offset values all 0. When I replug the scope, and restart the program, the offset errors are back again.

Steps to reproduce the behavior:
1. Start the program
2. Calibrate the scope by follow instructions in readme
3. Click Save settings
4. Click Exit
5. Replug scope
6. Start program
7. Observe that it still measures 60mV in the 20mV/div setting, with the probes shorted to GND; measures 110 mV in 200mV/div. For 500mV/div and up, the error is basically nothing.
8. When I now do the calibration procedure again, it's good up to when I restart the program again

```
$ cat .config/OpenHantek/DSO-6022BL_4541C616915B_calibration.ini
[eeprom]
replace_eeprom=false

[gain]
ch0\1000mV=1
ch0\100mV=1
ch0\2000mV=1
ch0\200mV=1
ch0\20mV=1
ch0\5000mV=1
ch0\500mV=1
ch0\50mV=1
ch1\1000mV=1
ch1\100mV=1
ch1\2000mV=1
ch1\200mV=1
ch1\20mV=1
ch1\5000mV=1
ch1\500mV=1
ch1\50mV=1

[offset]
ch0\1000mV=0
ch0\100mV=0
ch0\2000mV=0
ch0\200mV=0
ch0\20mV=0
ch0\5000mV=0
ch0\500mV=0
ch0\50mV=0
ch1\1000mV=0
ch1\100mV=0
ch1\2000mV=0
ch1\200mV=0
ch1\20mV=0
ch1\5000mV=0
ch1\500mV=0
ch1\50mV=0
```

**Expected behavior**

The offset setting should be saved correctly, so that they are remembered across program starts.

**Computer environment (please complete the following information):**

[image]

 - OpenHantek version: 3.4.0
 - OS: Arch Linux
 - Distribution, version: N/A, rolling
 - Screen resolution, screen setup (e.g. multimonitor, etc.): N/A
 - Video hardware: N/A
 - OpenGL version: Graphic: 4.6 (Compatibility Profile) Mesa 25.0.1-arch1.2 - GLSL version 1.20
 - Qt version: Qt 5.15.16
 - C++ compiler version: ?

**Scope device (please complete the following information):**
 - Device Hantek DSO-6022BL
 - Program top line: I don't have this on Arch Linux/i3; maybe it could be useful to log this to the console/terminal?
 - Input signal: grounded probes
 - Probe setting: probe 10X

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

redfast00, 2025-03-28:
Strangely enough, when I did the procedure in https://github.com/Ho-Ro/Hantek6022API/blob/main/README.md#create-calibration-values-for-openhantek, it does generate sensible gain/offset values, and I now use those.

Ho-Ro, 2025-04-01:
Maybe your initial offset was too big and the online calibration has stopped. You could try the folowing:
- Start OpenHantek, select two channels, timebase 10 ms/div.
- Stop OpenHantek (this prepares all settings without producing too much debug info in the next step).
- Start OpenHantek from command line with `OpenHantek --verbose 3`
- Select Menu/Oscilloscope/Calibrate Offset/Apply. A red label `OFFSET CALIBRATION` appears in the top left.
[image]
- Slowly cycle through all voltage/div settings for CH1 and CH2 (about 1 step per second).
- Select Menu/Oscilloscope/Calibrate Offset. The red label will disappear.
- Stop OpenHantek.

Scroll through the console output:

```
LANG=C ./OpenHantek --verbose 3
Version: 3.4.0-dirty - build: Mar 25 2025 09:13:46
0 ms: create openHantekApplication
...
1258 ms: create DSO control thread
 HantekDsoControl::HantekDsoControl()
  Calibration data: "DSO-6022BE_C3B36055284A_calibration.ini"
  HDC::getCalibrationFromEEPROM()
...
  Calibrate offset true
...                           <--- A lot of output when you switched the voltages 
  Calibrate offset false
  MainWindow::closeEvent()
 DsoSettings::save() "/home/horo/.config/OpenHantek/DSO-6022BE_C3B36055284A.conf"
380406 ms: application closed, clean up
  HDC::quitSampling()
  HDC::updateSamplerateLimits() 100 ... 1.5e+07
  HDC::quitSampling()
  HDC::updateSamplerateLimits() 100 ... 1.5e+07
 HDC::prepareForShutdown()
  Write calibration data into EEPROM
OpenHantek6022 has stopped after 382 s
...
```
Important is the line `Write calibration data into EEPROM` at the end.
If something went wrong (e.g. too big offset or jitter) you'll see this:

```
  Calibrate offset true
  Live calibration error
  Calibrate offset false
```

Scopes with the possibility to store offset values in EEPROM (e.g. 6022BE/BL) do not modify the `*.calibration.ini` file. This is only a fallback solution for (almost) compatible scopes (e.g. DDS120) that have a different firmware w/o EEPROM read/write support.

Ho-Ro, 2025-04-24:
@redfast could you follow my hints or shall we close as done?

redfast00, 2025-04-24:
Oops, my bad, I forgot. I don't have my scope with me right now, I'll try to get it done somewhere in the coming weeks.

KocsisV, 2025-06-07:
I attached a logic analyzer to the i2c lines, and the calibration data gets written into the eeprom. If I read back via an external reader the eeprom contains the correct (updated) values.

So either I'm doing something wrong during calibration or the data written out to the eeprom is incorrect. I have a 67mV (x1) offset on channel 1 after I restart the application and it re-reads the data from the eeprom.

redfast00, 2025-07-10:
No worries :) I'm not currently blocked by this, I can use the script workaround (and even then, I'm not entitled to your time anyways ;) )