OpenHantek/OpenHantek6022 #434: UI bugs out when using qt5ct/qt6ct

djsigmann, 2025-06-28:
When the environment variable `QT_QPA_PLATFORMTHEME` is set to either `qt5ct` or `qt6ct`, the application has some sort of weird UI behaviour, where the dialogs to tune scope functionality are stretched out and cannot be minimized..
This occurs both when connecting to a device or using the demo mode.

[image]

Setting `QT_QPA_PLATFORMTHEME` to an empty value does allow the application to run and function properly, albeit, without themes being applied.

This problem suddenly occurred the other day, checking my package manager logs, neither `OpenHantek6022` nor `qt6ct` were updated in the time between the last successful invocation of the program and the first time this bug occurred, nor were any QT-related settings changed.

Setup:
 - OpenHantek version: Latest commit
 - OS: Arch Linux
 - Screen resolution, 1920x1080
 - Video hardware: NVIDIA GeForce GTX 1060 6GB
 - OpenGL version: 4.6.0 NVIDIA 575.64
 - Qt version: 5.15.17+kde+r123-1
 - C++ compiler version: 15.1.1 20250425
 - Device: Hantek 6022BE

Ho-Ro, 2025-08-30:
Please check if the latest devdrop that uses Qt6 instead of Qt5 fixes your issues.

djsigmann, 2026-09-03:
This does indeed still occur on `OpenHantek6022 3.4.1-rc2`.