OpenHantek/OpenHantek6022 #439: MacOS Sequoia 15.6.1 Not Recognizing New 6022BL

davidbitton, 2025-09-05:
**Describe the bug**
My 6022BL is connected to a Thunderbolk dock (doesn't work direct connected either), and the red power lead is connected to a USB on my iMac. Nothing is show in the dropdown

**To Reproduce**
Launch app

Steps to reproduce the behavior:
1. launch app

**Expected behavior**
my 6022bl in the dropdown

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.4.0
 - OS: macos 15.6.1
 - Distribution, version: Sequoia
 - Screen resolution, screen setup (e.g. multimonitor, etc.): internal
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device 6022BL
 - Program top line OpenHantek6022 (3.4.0) - Demo Mode
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting:  probe X10
 - 
**Additional context**
I also can't get the Win11 driver to install in my Win11 VM. Windows says it can't find any driver in the folder.

davidbitton, 2025-09-05:
I also want to mention that the PID and VID are 0x3881 and 0x0925 respectively.

Ho-Ro, 2025-09-05:
Please toggle the switch on the back of the device and retry.

davidbitton, 2025-09-05:
> Please toggle the switch on the back of the device and retry.

Both positions, same thing. the device appears in System Information, but I'm not seeing the PID/VID values the app is looking for

[image]

davidbitton, 2025-09-05:
This PID/VID combo appears on line 168 in the `modelDSO6022.cpp` file from the latest SDK.

closed 2025-09-05 by davidbitton